mit-aera/FlightGoggles #17: Flight controller in AgileDrone

nominator, 2019-01-20:
Just a bit off-topic question, the two models of drones you used in research Penguin and Sparrow. Are you running the flight controller code on the Jetson TX1/TX2? Does it meet the hard real-time requirements for a flight controller?

Winter-Guerra, 2019-01-21:
For our ICRA 2018 paper, we ran the flight controller on the drone's TX1. The TX1 was running linux, so the flight controller was not guaranteed a hard rate. Instead, each loop was handled by the linux scheduler. We performed some CPU optimizations to reduce overall CPU load, to reduce scheduling time. We found that this approach was sufficient for agile maneuvers at up to ~2.86g on a TX2 (3rd version of drone: Blackbird).

closed 2019-01-21 by Winter-Guerra

nominator, 2019-01-21:
Thanks for the reply. This means that TX2 is quite capable of handling high maneuvering flight control. Good to know since it will simplify the hardware needed for flying a drone significantly.