mit-aera/FlightGoggles #23: Request Timeout During Catkin Build

BhavyanshM, 2019-02-09:
Hello, 

During the catkin build, flightgoggles package times out and fails the build with the following error:

Errors     << flightgoggles:make /home/sniper/catkin_ws/logs/flightgoggles/build.make.000.log   
CMake Error at /usr/share/cmake-3.10/Modules/ExternalData.cmake:1121 (message):
  

  Object MD5=7322a3165e02cc0c023336816df41b5d not found at:

    http://d34kgw45d3q5oc.cloudfront.net/Public/FlightGoggles/HashedReleases/MD5/7322a3165e02cc0c023336816df41b5d ("Timeout was reached")
    ftp://fast-nas.mit.edu/Public/FlightGoggles/HashedReleases/MD5/7322a3165e02cc0c023336816df41b5d ("Timeout was reached")

make[2]: *** [renderer_version/FlightGoggles_Data.tar.gz-hash-stamp] Error 1
make[1]: *** [CMakeFiles/flightgogglesBinaryDownload.dir/all] Error 2
make: *** [all] Error 2

varunmurali1, 2019-02-09:
Could you try getting either of these links using wget or by opening the link on a browser? For the first build, you should be connected to the internet to pull binaries.

strombom, 2019-02-09:
As a temporary work around I downloaded the binaries from the release:
https://github.com/mit-fast/FlightGoggles/releases

Comment out line 56-79 in CMakeLists.txt in src/flightgoggles/flightgoggles to be able to build without downloading.

BhavyanshM, 2019-02-09:
> As a temporary work around I downloaded the binaries from the release:
> https://github.com/mit-fast/FlightGoggles/releases
> 
> Comment out line 56-79 in CMakeLists.txt in src/flightgoggles/flightgoggles to be able to build without downloading.

Alright, and where does the binary need to be placed?

BhavyanshM, 2019-02-09:
> Could you try getting either of these links using wget or by opening the link on a browser? For the first build, you should be connected to the internet to pull binaries.

Yup, both wget and browser download the file just fine when connected.

strombom, 2019-02-09:
> Alright, and where does the binary need to be placed?

I simply put it in a separate folder and launched the executable before launching the teleopExample:
roslaunch flightgoggles teleopExample.launch

The simulator however does not reset properly after crashing so maybe it's better to install it as intended, not sure if that is related.

strombom, 2019-02-09:
If you unpack the release in src/flightgoggles/flightgoggles then you can start it with:
rosrun flightgoggles FlightGoggles.x86_64

BhavyanshM, 2019-02-09:
> I simply put it in a separate folder and launched the executable before launching the teleopExample:
> roslaunch flightgoggles teleopExample.launch
>
> The simulator however does not reset properly after [...]

Ahh, I see. Just not used to catkin a lot. Based on the comment "# Download FG executable to: ${CATKIN_INSTALL_PREFIX}/${CATKIN_PACKAGE_BIN_DESTINATION}", did you try placing the binary there? Also, what is the best way to find out the values of those variables?

BhavyanshM, 2019-02-09:
> If you unpack the release in src/flightgoggles/flightgoggles then you can start it with:
> rosrun flightgoggles FlightGoggles.x86_64

Did this fix the incorrect crash reset issue?

Winter-Guerra, 2019-02-09:
Hi all,

This timeout might be from slow internet. I'll look into relaxing Catkin's timeout parameter.

As a workaround, try something similar to the following:
```bash
mkdir -p catkin_ws/devel/lib/flightgoggles
cd catkin_ws/devel/lib/flightgoggles
wget http://d34kgw45d3q5oc.cloudfront.net/Public/FlightGoggles/StandaloneBinaries/FlightGoggles_Linux_v2.0.0.zip
gunzip FlightGoggles_Linux_v2.0.0.zip
chmod +x FlightGoggles_Linux_v2.0.0/FlightGoggles.x86_64
# Run the executable
rosrun flightgoggles FlightGoggles.x86_64
```

Winter-Guerra, 2019-02-09:
**Note:** Collisions do not reset the simulator and the simulator stops when the drone has collided. To fly around without colliders enabled for the purpose of exploring the environment, colliders can be disabled by setting `ignore_collisions` to true in the launch file or passing it as an argument to the launch file without editing the launch file.

E.g.
`roslaunch flightgoggles teleopExample.launch ignore_collisions:=1`

strombom, 2019-02-09:
> **Note:** Collisions do not reset the simulator and the simulator stops when the drone has collided. To fly around without colliders enabled for the purpose of exploring the environment, colliders [...]

Ok thank you, what is the "Reset" button on the bottom left for?

closed 2019-02-09 by Winter-Guerra

varunmurali1, 2019-02-09:
Unsure what "Reset" button you are referring to? I assume you mean the reset button in RVIZ in which case this is the function "The time panel also lets you reset the visualizer's internal time state -- this causes a reset of all the displays, as well as a reset of tf's internal cache of data."  per RVIZ user guide (http://wiki.ros.org/rviz/UserGuide). Although this resets RVIZ, it does not reset the simulator.

Winter-Guerra, 2019-02-09:
Hey all, I've pushed a patch that increases the CMake download timeout to 1 hour. You should be able to build as normal without workarounds. If you need more time, increase the timeout in `flightgoggles/CMakeLists.txt` line `47`.

BhavyanshM, 2019-02-09:
> Hey all, I've pushed a patch that increases the CMake download timeout to 1 hour. You should be able to build as normal without workarounds. If you need more time, increase the timeout in `flightgoggles/CMakeLists.txt` line `47`.

Thank you so much. The patch works fine.

commit a41a9a42 (2019-02-09): Increased binary download timeout to 1 hour. Closes #23.