mit-aera/FlightGoggles #25: Neither Keyboard Nor XBOX Controller Responding

BhavyanshM, 2019-02-09:
In the Teleop mode, even with the keyboard window open and spacebar long pressed, the drone does not move. Same with the Xbox One S controller.

Winter-Guerra, 2019-02-09:
Hi there, have you tried increasing the throttle using the keyboard
(Spacebar + W)? The drone will not respond to commands until extra throttle
is applied. You also must make sure to have the keyboard control window
selected before pressing any keys (see readme for a picture of the keyboard
control window).

BhavyanshM, 2019-02-09:
> Hi there, have you tried increasing the throttle using the keyboard (Spacebar + W)? The drone will not respond to commands until extra throttle is applied. You also must make sure to have the [...]

Oh I see. The problem is that since it has been running so slow, I had to long press Spacebar + W to see the effect. The drone is only somewhat controllable at that latency. How could it be that slow on Ryzen 7 2700x with GTX 1080 Ti? Anyhow, the Xbox controller does not respond at all.

BhavyanshM, 2019-02-09:
> > Hi there, have you tried increasing the throttle using the keyboard (Spacebar + W)? The drone will not respond to commands until extra throttle is applied. You also must make sure to have the keyboard control window selected before pressing any keys (see readme for a picture of the keyboard control window).
> > […](#)
> 
> Oh I see. The problem is that since it has been running so slow, I had to long press Spacebar + W to see the effect. The drone is only somewhat controllable at that latency. How could it be that slow on Ryzen 7 2700x with GTX 1080 Ti? Anyhow, the Xbox controller does not respond at all.

Is there anyway to reduce the quality to improve speed of simulation? Doesn't feel like a racing drone at all at the moment.

closed 2019-02-09 by BhavyanshM

Winter-Guerra, 2019-02-09:
Try doing a pull and run it again. I think that your teleop was running at a sub realtime rate. I turned on realtime simulation rate forcing in the teleop config.

BhavyanshM, 2019-02-09:
> Try doing a pull and run it again. I think that your teleop was running at a sub realtime rate. I turned on realtime simulation rate forcing in the teleop config.

The yaw, pitch and roll responses are now more realtime, but the throttle and gravity feel really unnatural. Also, is the Xbox controller not supported?

Winter-Guerra, 2019-02-09:
Keyboard control for throttle is not great because drones are not meant to
be controlled via keyboard. It feels a LOT better with joypad. Keyboard and
joypad are only meant for debugging, since this is an autonomous drone
challenge.

Xbox controller could work, but would need some button/axis remaps. Take a
look at the universal_teleop node and the different possible launch
configs.

We only support one controller type out of the box (Logitech F310) for the
moment. Might add Xbox in the future if there are enough requests. Perhaps
make a feature request GitHub issue or a pull request?

BhavyanshM, 2019-02-09:
> Keyboard control for throttle is not great because drones are not meant to be controlled via keyboard. It feels a LOT better with joypad. Keyboard and joypad are only meant for debugging, since this [...]

Ahh, you were right, the Xbox controller plays it a LOT better :smile: 

I got it working very well. The scale values tuned perfectly. Feels like a drone racing simulator now. Here is the pull request on the ros-universal-teleop: https://github.com/mit-fast/ros-universal-teleop/pull/1

You are welcome! By the way hats off to you guys.

Winter-Guerra, 2019-02-09:
> ros-universal-teleop: mit-fast/ros-universal-teleop#1

No problem! Thanks for the pull request! It has been merged.