mit-aera/FlightGoggles #26: minimum hardware required ?

MerouaneGuettache, 2019-02-09:
we don't have the same hardware performance describes on the README, though we prefer using our local machines rather than AWS instances.
what are the minimum hardware requirements to run the simulator ?

dineshresearch, 2019-02-09:
ya can you please reply even i need to run it in my local machine

paulerikf, 2019-02-09:
I have a 1060, the teleop example runs a bit slow for me, and the reporting modes run very slow.  That said, they still run, doesn't seem like a huge issue for me.  I'd give it a try and see for yourself.

Winter-Guerra, 2019-02-09:
Hi all,

FlightGoggles is similar to the Gazebo simulator in that it can scale the ROS clock down if the simulation is running slowly. Thus, your autonomous algorithms will see a constant 60Hz camera in "sim time" and should experience accurate drone dynamics as long as the camera renderer is able to run (see http://wiki.ros.org/Clock).

At the moment, FlightGoggles requires ~`3.4GB` of VRAM and has the same underlying requirements as the Vulkan rendering API (  https://en.wikipedia.org/wiki/Vulkan_(API)  ).

I'd try to run the simulation environment on your local machine. If that does not work, then you could try running it on AWS. It does not really matter if your computer runs the simulation slowly, any algorithmic evaluation would happen in "sim time" in the cloud.

Best,
-Winter

closed 2019-02-10 by Winter-Guerra