mit-aera/FlightGoggles #27: RViz says "No image"

nilay994, 2019-02-09:
RViz image window doesn't show any images. The topics window has a warning saying that no image is published in `bounding_box/rgb`.   Maybe I missed something while performing the steps mentioned in the README of the master branch of this repository. 

How are the scenes rendered to the image window when trying to simulate locally, are we supposed to download some scene files before running the ros.launch files? 

Secondly, the README does mention about AWS, is it a mandatory step while getting FlightGoggles to run? Or is it a provision for users in case the local computer fails to have a good graphic card?

varunmurali1, 2019-02-09:
Can you rostopic echo /uav/camera/left/image_rect_color and /uav/camera/left/ir_beacons and report if you have any messages on those topics?

Winter-Guerra, 2019-02-09:
@nilay994 do you see any errors when you run `roslaunch flightgoggles teleopExample.launch` or `rosrun flightgoggles FlightGoggles.x86_64`?

Winter-Guerra, 2019-02-10:
@nilay994, are you still having this problem?

nilay994, 2019-02-10:
@Winter-Guerra , thank you for checking on this opened issue. To narrow down my problem, I am currently setting up [Bumblebee](https://wiki.ubuntu.com/Bumblebee) since I use a laptop with mobile graphics (Quadro P1000). I am suspecting it is the rendering speed. I should close this issue for now.

closed 2019-02-10 by nilay994

nilay994, 2019-02-14:
As far as I know, it seems to be a problem with my graphics. My current environment can't adjust the brightness or switch to nvidia properly - maybe the renderer flushes to `Screen 0` by default, and my computer doesn't know what Screen 0 is. I have a hybrid graphics setup, Bumblebee didn't work for me yet. I'll be running FG on some other computer to get the simulator to run.