mit-aera/FlightGoggles #28: FlightGoggles Renderer error and exits with teleopExample.launch

MattAxG, 2019-02-09:
When launching via 
_roslaunch flightgoggles teleopExample.launch_
or with 
_roslaunch 'flightgoggles teleopExample.launch use_external_renderer:=1_

... the renderer will appear but close almost immediately. 

The following message is returned.
_Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application._

When launching without the _use_external_renderer:=1_ argument you can also see the following messge

[uav/flightgogglesRenderer-6] process has died [pid 19209, exit code -6, cmd /home/myws/catkin_ws/devel/lib/flightgoggles/FlightGoggles.x86_64 __name:=flightgogglesRenderer __log:=/home/myws/.ros/log/a8fef8a0-2c98-11e9-90f2-002522d2019b/uav-flightgogglesRenderer-6.log].
log file: /home/myws/.ros/log/a8fef8a0-2c98-11e9-90f2-002522d2019b/uav-flightgogglesRenderer-6*.log

When I dug into the log file it pointed me to another log (Player.log) in the unity3d.
[Player.log](https://github.com/mit-fast/FlightGoggles/files/2848122/Player.log)

Possibly related:
When I attempt to run the _roslaunch flightgoggles core.launch_ instead I get similar behavior to the other ticket "RViz says "No image" #27"
The renderer (FlightGoggles unity window) stays open but continues to show the "waiting for connection from client" 
The third_person_drone_follower.rviz window continues to show "No Image"
rostopic echo for /uav/camera/left/image_rect_color and /uav/camera/left/ir_beacons show no messages for those topics.

Winter-Guerra, 2019-02-09:
This crash seems to be unity related. Could you please post the unity player log?

MattAxG, 2019-02-09:
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2018.3.0f2 (6e9a27477296)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: count=64
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance3, enabled=0
[Vulkan init] extensions: name=VK_KHR_multiview, enabled=0
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_relaxed_block_layout, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_sampler_ycbcr_conversion, enabled=0
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_storage_buffer_storage_class, enabled=0
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_variable_pointers, enabled=0
[Vulkan init] extensions: name=VK_EXT_blend_operation_advanced, enabled=0
[Vulkan init] extensions: name=VK_EXT_conservative_rasterization, enabled=0
[Vulkan init] extensions: name=VK_EXT_depth_range_unrestricted, enabled=0
[... 1111 lines cut ...]
(Filename:  Line: 803)

Receiving unhandled NULL exception
#0  0x0000000095ff30 in std::pow(float, float)
#1  0x00000000960209 in std::pow(float, float)
#2  0x00000000aee4f6 in std::pow(float, float)
#3  0x0000000095516b in std::pow(float, float)
#4  0x00000000cfb5ef in std::pow(float, float)
#5  0x00000000b32188 in std::pow(float, float)
#6  0x00000000cd487a in std::pow(float, float)
#7  0x00000000abb8e3 in std::pow(float, float)
#8  0x00000000ac2a0a in std::pow(float, float)
#9  0x00000000bd7d1c in std::pow(float, float)
#10 0x00000000bd8acf in std::pow(float, float)
#11 0x0000000096c27f in std::pow(float, float)
#12 0x00000000bf3581 in std::pow(float, float)
#13 0x00000000bf3ef8 in std::pow(float, float)
#14 0x00000000bef3a7 in std::pow(float, float)
#15 0x00000000cf768e in std::pow(float, float)
#16 0x000000009168ee in std::pow(float, float)
#17 0x00000000910d92 in std::pow(float, float)
#18 0x00000000910ec9 in std::pow(float, float)
#19 0x00000000ca3970 in std::pow(float, float)
#20 0x007f97c57b86ba in start_thread
#21 0x007f97c54ee41d in clone

Winter-Guerra, 2019-02-09:
Looks like something is going NULL in the renderer. Does `vulkan-smoketest` pass? If not, could you please post your NVidia driver version?

MattAxG, 2019-02-09:
I just ran it and it look successful.  (unless I needed to let it run longer)

here is the output (besides the funny teapot image window that is)
```
303 presents in 5.00537 seconds (FPS: 60.5349)
298 presents in 5.00307 seconds (FPS: 59.5634)
302 presents in 5.01337 seconds (FPS: 60.239)
298 presents in 5.00035 seconds (FPS: 59.5958)
301 presents in 5.0166 seconds (FPS: 60.0008)
301 presents in 5.01646 seconds (FPS: 60.0024)
300 presents in 5.00006 seconds (FPS: 59.9993)
300 presents in 5.00023 seconds (FPS: 59.9972)
301 presents in 5.01644 seconds (FPS: 60.0027)
300 presents in 5.00022 seconds (FPS: 59.9974)
301 presents in 5.01642 seconds (FPS: 60.0029)
300 presents in 5.00003 seconds (FPS: 59.9997)
300 presents in 5.01669 seconds (FPS: 59.8004)
300 presents in 5.00036 seconds (FPS: 59.9957)
301 presents in 5.0163 seconds (FPS: 60.0044)
300 presents in 5.00009 seconds (FPS: 59.999)
301 presents in 5.01665 seconds (FPS: 60.0002)
301 presents in 5.01663 seconds (FPS: 60.0005)
300 presents in 5.00002 seconds (FPS: 59.9997)
301 presents in 5.01654 seconds (FPS: 60.0015)
^C

```

MattAxG, 2019-02-09:
Nvidia Driver info:
NVIDIA-SMI 410.79       Driver Version: 410.79       CUDA Version: 10.0

varunmurali1, 2019-02-09:
Are you using Ubuntu 16.04? Which GPU are you using?

MattAxG, 2019-02-09:
Description:	Ubuntu 16.04.5 LTS
Release:	16.04

Geforce RTX 2070

Winter-Guerra, 2019-02-09:
Hi @MattAxG,

Sorry for the wait. This is actually a non-trivial issue to solve. We're trying to replicate the issue on our side so that we can provide you with a solution.

Our current hypothesis is that you have a specific nvidia driver/vulkan API combo that has a hidden bug. We've got the following combinations to work:

Ubuntu: 16.04
Nvidia Driver: `410.48` 
Vulkan API: `1.0.61`

and 

Nvidia Driver: `384.130`
Vulkan API: `1.1.73`

Best,
-Winter

Winter-Guerra, 2019-02-09:
@MattAxG can you verify your FlightGoggles version from the splash screen? It should say `v2.0.0` or perhaps `v2.0.0-b5`.

MattAxG, 2019-02-09:
v2.0.0

Winter-Guerra, 2019-02-09:
OK, we're going to try to isolate a few variables. Could you please try downloading [this](http://d34kgw45d3q5oc.cloudfront.net/Public/FlightGoggles/DebugBuilds/v2.0.0-debug1.zip) linux binary and running it? It will not connect to ROS, but it will tell us if the renderer is broken or the ROS interconnect. Note, it might take a while to load.

MattAxG, 2019-02-10:
Done. 

Captured the terminal output here:
[terminal_output.txt](https://github.com/mit-fast/FlightGoggles/files/2848400/terminal_output.txt)

Here is the 
[Player.log](https://github.com/mit-fast/FlightGoggles/files/2848401/Player.log)

There was also some scroll in the terminal that might be useful here
[terminal_dump.txt](https://github.com/mit-fast/FlightGoggles/files/2848404/terminal_dump.txt)

varunmurali1, 2019-02-10:
Were you able to see an image with this executable? Were you able to look around with the mouse? It’s hard to tell from the output whether you were able to see the scene.

MattAxG, 2019-02-10:
That was the first run, it behaved very similar to the v2.0.0 release, open for a few seconds then closed.

I did run it again by double clicking the icon in the file manager instead of launching by command line. This gave me about 4 minutes of "flying around" before the window closed.

MattAxG, 2019-02-10:
Here is the Player.log for the longer run.
[Player.log](https://github.com/mit-fast/FlightGoggles/files/2848421/Player.log)

MattAxG, 2019-02-10:
Do you think it is related to either 

[V2] [Unity 2017.4] Post-Processing Stack causes crash on Linux with latest Nvidia driver #592
`https://github.com/Unity-Technologies/PostProcessing/issues/592`
or
NVIDIA bug 2254861

Winter-Guerra, 2019-02-10:
Hi @MattAxG,

This issue is a bit odd. With the debug build, you have been able to run the Unity3D backend for up to 4 minutes before crashing. This suggests that the error is coming from some hardware/software/driver interaction specific to your OS install or development environment setup. The difference between running the executable from command line and double clicking the executable is also concerning. Perhaps something odd is happening in your `.bashrc`? 

In the `player.log`, the warnings about `uav bindings` and `shader unsupported` are normal to the Unity3D Linux Vulkan renderer. Vulkan is refusing to render specific effects/shaders that are unsupported in Linux. Everyone running the binary has similar warnings in their logfiles.

We're going to build another set of binaries for you using a different version of Unity3D to see if that fixes the issue. Stay tuned! In the meantime, you might want to install this framework on another computer or hard drive. It is very possible that another computer/OS install would work.

Best,
-Winter

Winter-Guerra, 2019-02-10:
Sadly @MattAxG, I don't think that the issues you linked apply to this issue. We are using unity3d version `2018.3.0f2`, which is a lot newer than that of the issues linked.

Winter-Guerra, 2019-02-11:
Hi @MattAxG,

We were able to reproduce an error similar to your original error using Unity3D version `2018.3.5f1`. After some investigation, we think that the culprit is Unity's experimental ["Graphics Jobs" rendering backend](https://unity3d.com/learn/tutorials/topics/best-practices/multithreaded-rendering-graphics-jobs#Graphics%20Jobs). We've generated a new binary with "Graphics Jobs" disabled that seems to fix the problem for our test setup. Could you please try this executable and report back?

[Executable link](http://d34kgw45d3q5oc.cloudfront.net/Public/FlightGoggles/StandaloneBinaries/FlightGoggles_Linux_v2.0.0-no_multi_thread.zip)

To run this with the existing ROS framework, please run the downloaded binary and then run `roslaunch flightgoggles teleopExample.launch use_external_renderer:=1`

Best,
-Winter

juanmed, 2019-02-12:
Hello. I had the same issue @MattAxG  is describing. I jus tried the binary you provided in the last comment and now I am able to get reliable rendering for more than 4 minutes. I have been running it for 20minutes and seems to have no problem. Input from keyboard is working.

My system is:
Intel® Core™ i5-8400 CPU @ 2.80GHz × 6 
GeForce GTX 1050 Ti/PCIe/SSE2
NVIDIA-SMI 396.37 Driver Version: 396.37
Ubuntu 16.04.5 LTS

Vulkan smoke test passes just as described by @MattAxG. rostopic echo for /uav/camera/left/image_rect_color and /uav/camera/left/ir_beacons is now publishing messages for each.

I should mention that I have Unity3D Version 2018.3.0f2 (6e9a27477296) Personal installed in my computer. Since it seems to be an issue related to Unity I though this info might be helpful.

Winter-Guerra, 2019-02-12:
Wonderful, thanks for the feedback Juan!

koolmoo, 2019-02-12:
Hi, I am experiencing the same issue with the following system:

Intel Core i7-4790K
GeForce GTX 980 Ti
NVIDIA-SMI 384.130
Ubuntu 16.04.5 LTS
Vulkan API 1.0.57

I will try the new binaries tomorrow when I get the chance. Just wondering, would upgrading Vulkan to 1.1.73 help, or is the cause of the problem not related to Vulkan? I believe I'm on Unity3D Version 2018.3.0f2 just like @juanmed.

Also, I noticed in my player log that my Vulkan renderer is set to [Intel(R) Haswell Desktop] instead of my GTX 980 Ti. Do I have to configure it to use my GPU somehow?

Thanks!

varunmurali1, 2019-02-12:
I don't believe this issue was related to Vulcan. The player log suggesting that the Vulcan renderer is set to intel sounds concerning. Are you sure `nvidia-smi` is picking up the gpu and is not erroring out? This is also an old Nvidia driver and your issue could be possibly related to [this](https://bugs.launchpad.net/ubuntu/+source/nvidia-graphics-drivers-384/+bug/1726809). Would it be possible to update the driver to a newer version?

koolmoo, 2019-02-12:
Hi @varunmurali1, yes I am sure nvidia-smi sees the GPU. I can upgrade to a newer driver, but could you please suggest a version that is known to work? Thanks.

varunmurali1, 2019-02-12:
I am currently using nvidia-410. Could you please try that version?

yonghee93, 2019-02-12:
Hi I have same issue.

My system is:
Intel® Core™ i5-4690 CPU @ 3.50GHz × 4 
GeForce GTX 750 Ti/PCIe/SSE2
NVIDIA-SMI 384.130
Vulkan API version 1.0.49
Ubuntu 16.04.5 LTS

Vulkan smoke test passes well.
In Player.log, Vulkan renderer=[GeForce GTX 750 Ti] id=[1380].

Do I have to upgrade nvidia driver or GPU?

varunmurali1, 2019-02-12:
I suspect that a GTX 750 Ti will not be able to successfully render using FG. Typically, you want to use a GPU with >= 4 Gb VRAM. We are considering the possibility of releasing a version that will be able to render on less capable GPUs by sacrificing quality. Please stay tuned for this!

MattAxG, 2019-02-13:
I have been working with this version for a few hours. Seems to be perfectly stable. Thank you for the quick help.

varunmurali1, 2019-02-13:
That’s fantastic to hear! Glad it works for you.

Winter-Guerra, 2019-02-13:
@koolmoo have you had any luck with the new non-multithreaded binaries or with the new NVidia driver?

koolmoo, 2019-02-13:
@Winter-Guerra I updated the drivers to 410.78, but I'm not sure how to run the new binaries. Do I just run the FlightGoggles.x86_64 executable from wherever I unzipped it, or do I need to copy all the contents to catkin_ws/devel/lib/flightgoggles and overwrite the existing files?

I tried just running the executable, but the program closed as soon as it opened. Then I ran `roslaunch flightgoggles teleopExample.launch use_external_renderer:=1` but that gave the same result as before (no image).

Let me know how I should proceed. Thanks!

Winter-Guerra, 2019-02-13:
Looks like it is crashing on startup @koolmoo. Could you please post the player.log of the new executable? The location of the log is in the terminal printout. You can run it from anywhere and then use the same roslaunch command you used above.

jeffcyu, 2019-02-13:
Hi all I'm having the similar issue as @koolmoo with a crash at startup.

My setup is:
Intel Core i7-7700HQ
Nvidia GTX 1060 6gb
Xubuntu 16.04.5 LTS / ROS Kinetic

I've tried Nvidia drivers 384.130 and 410.78 (`nivia-smi` sees GPU in both cases) along with Vulkan API 1.0.57 using the unity binaries from the repo and the one posted in here, but the results are the same.

The output I get is:
`XIO:  fatal IO error 0 (Success) on X server ":0.0"
      after 225 requests (221 known processed) with 22 events remaining.`
`XIO:  fatal IO error 0 (Success) on X server ":0.0"
      after 225 requests (221 known processed) with 22 events remaining.
`
It seems like from looking at the player log that the issue is the renderer being set to the Intel integrated graphics rather than the GPU. 
[Player.log](https://github.com/mit-fast/FlightGoggles/files/2862526/Player.log)
Any thoughts on pointing Vulkan at the GPU?
Thanks for your time and efforts.

koolmoo, 2019-02-14:
Yeah I definitely think @jeffcyu and I have the same problem. Here is my player log for reference:

[Player.log](https://github.com/mit-fast/FlightGoggles/files/2862889/Player.log)

Thanks!

varunmurali1, 2019-02-14:
@jeffcyu @koolmoo I noticed something from both your player logs, you appear to be using the multi-threaded unity executable, not the [single threaded version](http://d34kgw45d3q5oc.cloudfront.net/Public/FlightGoggles/StandaloneBinaries/FlightGoggles_Linux_v2.0.0-no_multi_thread.zip). Can you run the new executable after setting the environment variable `VK_ICD_FILENAMES` to `/usr/share/vulkan/icd.d/nvidia_icd.json` as described [here](https://wiki.archlinux.org/index.php/Vulkan). 

Since this issue appears to be unrelated to the original problem with multi-threading could I suggest that we open a new issue with Vulkan failing to select the GPU but instead defaulting to the Intel icd?

yonghee93, 2019-02-14:
Hello, I had same issue.
After changing my gpu from GTX 750 ti to GTX 1050 ti,
It works well now.

koolmoo, 2019-02-14:
@varunmurali1 Hey that did it! After setting that environment variable and running the new binaries, the FlightGoggles simulator opens and shows a "waiting for client" screen. My player log now says that the Vulkan renderer is my 980 Ti GPU. Also, running the external renderer command opened ROS with the simulator image displayed. Thanks so much!

Quick side question: I heard there was a way to use Xbox controllers for teleoperation, but is it possible to connect an RC transmitter (e.g. FrSky Taranis) to control the drone?

varunmurali1, 2019-02-14:
@yonghee93  @koolmoo  glad it works for you.  

@koolmoo At this time, RC transmitters are not supported, sorry.

jeffcyu, 2019-02-14:
The fix with the new binary works! Thanks @varunmurali1

varunmurali1, 2019-02-14:
Great! Glad it works!

closed 2019-02-14 by varunmurali1

juanmed, 2019-02-15:
Hello, 

Previously I reported I had the same issue as @MattAxG. Using the v2.0.0_no_multithread binary I am able to connect and fly the drone ONLY when launching teleopExample.launch. When I try to launch core.launch there seems to be an issue and RViz keeps display "No Image" indefinitely.

This repeatable. While the v2.0.0_no_multithread binary is running, if I relaunch and close teleopExample.launch it alwasy connects and works, whereas the same process with core.launch never works and displays "No image".
 
This are my log files: 
Unity: [Player.log](https://github.com/mit-fast/FlightGoggles/files/2868684/Player.log)
teleopExample: [79e2774c-3105-11e9-a566-e0d55eb54399.tar.gz](https://github.com/mit-fast/FlightGoggles/files/2868691/79e2774c-3105-11e9-a566-e0d55eb54399.tar.gz)
core: [4a8e1636-3105-11e9-a566-e0d55eb54399.tar.gz](https://github.com/mit-fast/FlightGoggles/files/2868695/4a8e1636-3105-11e9-a566-e0d55eb54399.tar.gz)

I am not sure if this is the correct place to write this. Let me know if I should open an new issue.

Winter-Guerra, 2019-02-15:
Hi @juanmed, it looks like the dynamics and render engine were waiting for control input before starting. Sorry about that, I've pushed a patch that fixes this issue a few days ago. Here are some commands that should get you up and running.
```bash
cd ~/catkin_ws/src
wstool update
roslaunch flightgoggles core.launch
```

In the future, please make a new issue for new issues. This is not the correct thread for this.

juanmed, 2019-02-16:
Hi. Thanks for the indications. Also, I found it is necessary to rebuild the workspace in order for it to work, so the commands that worked for me were:

`cd ~/catkin_ws/src
wstool update
cd ~/catkin_ws/src/flightgoggles
catkin clean flightgoggles
catkin build
roslaunch flightgoggles core.launch`

After this both core.launch and teleopExample.launch were working.

Thanks.

vyerneni17, 2019-02-16:
> Hi @MattAxG,
>
> We were able to reproduce an error similar to your original error using Unity3D version `2018.3.5f1`. After some investigation, we think that the culprit is Unity's experimental [...]

I have tried downloading this executable and running it, below is the output

```Native stacktrace:

        /home/ubuntu/catkin_ws/FlightGoggles_Linux_v2.0.0-no_multi_thread/FlightGoggles_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so(+0xd5238) [0x7fb0e285b238]
        /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7fb0e55e0390]
        /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x38) [0x7fb0e523a428]
        /lib/x86_64-linux-gnu/libc.so.6(abort+0x16a) [0x7fb0e523c02a]
        ./FlightGoggles.x86_64() [0xd21ea6]
        /home/ubuntu/catkin_ws/FlightGoggles_Linux_v2.0.0-no_multi_thread/FlightGoggles_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so(+0x14e8b2) [0x7fb0e28d48b2]
        /home/ubuntu/catkin_ws/FlightGoggles_Linux_v2.0.0-no_multi_thread/FlightGoggles_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so(+0x5be9c) [0x7fb0e27e1e9c]
        /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7fb0e55e0390]
        ./FlightGoggles.x86_64() [0x954160]
        ./FlightGoggles.x86_64() [0x97f3f8]
        ./FlightGoggles.x86_64() [0x97f6fe]
        ./FlightGoggles.x86_64() [0x905d7f]
        ./FlightGoggles.x86_64() [0xcd95c2]
        ./FlightGoggles.x86_64() [0xcdcbc2]
        ./FlightGoggles.x86_64() [0xc31b75]
        ./FlightGoggles.x86_64() [0xc44613]
        ./FlightGoggles.x86_64() [0x44d95e]
        /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7fb0e5225830]
        ./FlightGoggles.x86_64() [0x45a431]

Debug info from gdb:

mono_gdb_render_native_backtraces not supported on this platform, unable to find gdb or lldb

=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries
used by your application.
=================================================================

Aborted (core dumped)```

benkmoore, 2019-03-01:
Hi,

I'm just getting started with Flgihtgoggles.
After running the new executable, I'm still getting the same error;

`[uav/flightgogglesRenderer-3] process has died [pid 3665, exit code 1, cmd /home/ben/catkin_ws/devel/lib/flightgoggles/FlightGoggles.x86_64 -screen-quality Fastest __name:=flightgogglesRenderer __log:=/home/ben/.ros/log/454dabbc-3c4c-11e9-b1bd-dc8b285dbe07/uav-flightgogglesRenderer-3.log].
log file: /home/ben/.ros/log/454dabbc-3c4c-11e9-b1bd-dc8b285dbe07/uav-flightgogglesRenderer-3*.log`

My setup is;
Intel® Core™ i7-7700HQ CPU @ 2.80GHz × 8 
Intel® HD Graphics 630 (Kaby Lake GT2) 
GP107M [GeForce GTX 1050 Ti Mobile]
Xubuntu 16.04.5 LTS / ROS Kinetic

The Player.log like produced says the driver is incompatible but I don't have clue how to go about fixing that or if that even is the issue.

`Desktop is 3840 x 2160 @ 60 Hz
[Vulkan init] extensions: count=1
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
Vulkan error VK_ERROR_INCOMPATIBLE_DRIVER (-9) file: ./Runtime/GfxDevice/vulkan/VKContext.cpp, line: 332
Vulkan detection: 0
No supported renderers found, exiting

(Filename:  Line: 634)

~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
~                                                                                                    
"Player.log" 9L, 316C`

Any help would be great, thanks

venky-intel, 2019-03-01:
@benkmoore What is the driver version you're using? You can check that using `nvidia-smi`

Also, make sure it satisfies the requirements mentioned [here](https://github.com/mit-fast/FlightGoggles/wiki/Prerequisites-and-Testing-Setup)

benkmoore, 2019-03-01:
Hey @venky-intel,

3D controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile]

I can't fin my exact graphics card in the vulkan compatibility list but the GeForce GTX 750 Ti is closest to it and is compatible so if that is ok then I think my system meets the requirements.

In additional drivers section of Software & updates I'm currently using:
`X.Org X Server -- Nouveau display driver from xserver-xorg-video-nouveau (open source)`

But it also has an option to select NVIDIA binary driver - version 384.130

Any advice? cheers

Winter-Guerra, 2019-03-02:
Hi @benkmoore,

Please try using the Nvidia binary driver. The open-source driver is not distributed by Nvidia and is generally incompatible with high-end 3D graphics rendering. Please also make sure that you are using flightgoggles renderer binary versions v2.0.3 or v2.0.4-b1 and have installed `libvulkan1`.

If you are still having trouble, please open a new issue. Your issue does not seem related to this thread.

Thanks,
-Winter