mit-aera/FlightGoggles #30: Support for 18.04 LTS

abhijit-15, 2019-02-10:
Any work-arounds to install on Ubuntu18.04 LTS? Also shouldn't the latest LTS release also be supported? Thanks!

BhavyanshM, 2019-02-10:
> Any work-arounds to install on Ubuntu18.04 LTS? Also shouldn't the latest LTS release also be supported? Thanks! I understand that its tested and built on ROS Kinetic but would like to know if there is a way to use FlightGoggles with ROS Melodic?

Works perfectly fine on Ubuntu 18.04 running ROS Melodic for me.

Winter-Guerra, 2019-02-10:
The simulation framework probably will work in 18.04, but is not officially supported. Kinetic does not support 18.04. We are only supporting Kinetic at the moment for simplicity.

closed 2019-02-10 by Winter-Guerra

abhijit-15, 2019-02-10:
Thank you so much. It works! Had some other dependency issues that i had to fix. You guys are the best!

Xiad32, 2019-02-16:
@abhijit-15 can you report the "other dependencies" it required here for everyone else. Thanks

matthew-t-watson, 2020-01-22:
> Thank you so much. It works! Had some other dependency issues that i had to fix. You guys are the best!

Could you elaborate on the dependency issue you fixed?

isro01, 2020-01-25:
write melodic instead of kinetic in the official instructions . and in the inertialMeasurementSim.hpp file in 
src/flightgoggles/flightgoggles_uav_dynamics/libs/multicopterDynamicsSim add #include<random>

seems to work then

vlsalsa, 2020-02-05:
I attempted to install melodic instead using the following command:
`rosdep install --from-paths src --ignore-src --rosdistro melodic -y`

Instead, I get:

> error: the following packages/stacks could not have their rosdep keys resolved to system dependencies: 

> keyboard: No definition of [ruby] for OS version 

Tried an install of Ruby rails but no dice... same error.