mit-aera/FlightGoggles #34: Missing include `unistd.h` in `generalclient.hpp`

foehnph, 2019-02-11:
Hi and thanks for releasing FlightGoggles

I met an issue while compiling on Ubuntu18.01 with ROS Melodic.
`flightgoggles/flightgoggles_ros_bridge/src/GeneralClient/GeneralClient.cpp` uses `sleep()` and `usleep()` but neither the header nor the source include it.
It is easily resolved by including `unistd.h` in the source (preferred) or the header.

Shall I do a pull request or do you fix it quickly by committing? 
Thanks a lot!

Winter-Guerra, 2019-02-11:
Hi Philipp,

Sorry about that. A pull request would be appreciated!

Thanks,
-Winter

BhavyanshM, 2019-02-12:
> Hi Philipp, Sorry about that. A pull request would be appreciated! Thanks, -Winter
> […](#)

Here @Winter-Guerra https://github.com/mit-fast/FlightGoggles/pull/39

foehnph, 2019-02-12:
@BhavyanshM 
@Winter-Guerra 

thanks a lot!

Winter-Guerra, 2019-02-12:
PR merged. Thanks @BhavyanshM and @foehnph for catching this issue! We're running 16.04, so we didn't see this issue before. GCC in 16.04 seems to automagically include the file.

closed 2019-02-12 by Winter-Guerra