mit-aera/FlightGoggles #42: no effect of control_nodes/joy on the simulator

nilay994, 2019-02-14:
I use `gstest-gtk` to check the axes of my non-xbox controller. I then remap the axes in the `example_input.yml` file. I also uncommented all the joy_buttons, keys and scale options while doing this.  After I make the yml file changes, I run `catkin build` in the `catkin_ws` directory, I use `roslaunch flightgoggles teleopExample.launch` to start the simulator. 
When I echo the messages from `control_nodes/joy`, I see that the joystick axis events are being generated according to the jstest-gtk calibration. However, nothing moves on the simulator. I just see the stationary view from the drone. If I try the keyboard at this moment, it works perfectly. Just that the joy messages don't seem to reach the simuator. 

Any suggestions on this? Am I missing any controller remapping steps?

Winter-Guerra, 2019-02-14:
Hi @nilay994,

Joypads that are not the Logitech F310 and Xbox One controller are not officially supported. However, I think that you are on the right track. To activate controller control, the "override" button must be held on the controller. This button mapping is defined in the "keys" input mapping in the yaml. If the controller inputs are mapped correctly, you should see activity on the `/uav/input/rateThrust` channel when the controller override is enabled.

Best,
-Winter

Winter-Guerra, 2019-02-16:
@nilay994, are you still having this issue?

nilay994, 2019-02-16:
@Winter-Guerra , No I haven't gotten it to work yet. I don't understand the difference between the fields `buttons` and `keys` in the `example_input.yaml`. I did try to map the `override` field under keys and buttons of the yaml to the joystick buttons/keys - but without any luck. I see that the `/control_nodes/joy` is publishing correctly but nothing is published on `/uav/input/rateThrust`.

Winter-Guerra, 2019-02-17:
Hi @nilay994,

After closer inspection, it looks like you need to change the `override` parameter in the `joy_buttons` param and maybe the axis mappings in the `joy_axes` param. You can find these mappings experimentally by looking at the `/control_nodes/joy` topic. 

`keys` refers to remaps for the keyboard controller and should not be uncommented. I'm terribly sorry about the naming confusion.

The default values and structure of the joystick params can be found in lines 23-51 of [this file](https://github.com/mit-fast/ros-universal-teleop/blob/master/src/teleop.cpp#L23). I'd advise only defining values for `joy_buttons` and `joy_axes` in your remapping file for simplicity. If any of your axes are flipped, I'd then advise defining `axis_scales` values to fix the flipped axes.

Best of luck!
-Winter

nilay994, 2019-02-18:
hi @Winter-Guerra , thank you for your reply. I still haven't gotten it to work. I tried it with a PS3 controller and FrSky controller. For now, I use the keyboard. I'll close the issue, since I see that others were successful in configuring it. I will comment on the closed issue in case I realize the mistake or in case I still feel stuck.

closed 2019-02-18 by nilay994