mit-aera/FlightGoggles #43: Collision simulation

ashwinvk94, 2019-02-14:
The flight goggles simulator seems to shutdown when the drone crashes into an obstacle. Is there any way to  reset the quad to the start position without having to run the launch file from start again?

Winter-Guerra, 2019-02-14:
Hi @ashwinvk94,

At the moment, restarting the launch file is the only way to restart the simulation after a collision. If you'd like to do tests without collisions enabled, it is possible to disable them. We're looking into making an easier way of restarting the simulation in the future, but for right now restarting the launch file or disabling collisions is the only way.

Best,
-Winter

ashwinvk94, 2019-02-14:
Got it thanks.

PHRABAL, 2019-02-15:
@Winter-Guerra I've disabled collisions, the drone starts to fly, then I want to reset it. Is /uav/input/reset topic working? I've tried a number of ways to publish to it and nothing seems to reset it (by which I mean put the drone back to the starting position on the ground). 

For example, I've tried amongst others:

`rostopic pub /uav/input/reset std_msgs/Empty "{}"`

Also, is there a way to set the position of the drone? Like through the message `geometry_msgs/Pose`?

Winter-Guerra, 2019-02-15:
Hi @PHRABAL, restarting the dynamics node is currently the fastest way to reset the drone to the start position. However, we accept pull requests!

Winter-Guerra, 2019-02-17:
Hi all,

We have an initial implementation of a quick reset function. In our implementation the drone will reset position on every collision. We're currently testing the patch and will release as soon as possible.

Best,
-Winter

PHRABAL, 2019-02-17:
@Winter-Guerra great. In the meantime, I implemented a function to reset the env which does a rostopic kill combined with setting those topics to respawn=True in the launch file. It works. Takes about 1-2 seconds to complete the loop on a p3.

Winter-Guerra, 2019-02-17:
The new patch is out! FlightGoggles should reset the drone to the start position after a collision.

```
# Pull new flightgoggles source code
cd ~/catkin_ws/src/flightgoggles
git pull
# Force flightgoggles to redownload binary at build time
catkin clean
catkin build
```

PHRABAL, 2019-02-17:
@Winter-Guerra Thanks for the quick add of this function. If we want to reset it manually other than a collision, is there a way to do that? Sometimes the drone gets "stuck" without registering an actual collision.

eatal, 2019-02-17:
Hi @PHRABAL ,
You can reset the dynamics by publishing an empty message `std_msgs/Empty` to the ropic `/uav/collision`. E.g. from a terminal using the command:
`rostopic pub /uav/collision std_msgs/Empty`

Note that there is a timeout after each reset, during which the drone cannot go to armed state (similar to when the simulation is initialized and no thrust commands have been given yet). The timeout is set to 0.1 seconds by default, but you can change this in `src/FlightGoggles/flightgoggles/config/drone/drone.yaml`where the parameter is listed as `reset_timeout`.

Hope this helps,
-Ezra

PHRABAL, 2019-02-17:
@eatal you made my day ... maybe week.

Winter-Guerra, 2019-02-18:
Closing since this feature request has been implemented. Instructions for manual reset will be added to the README FAQ.

closed 2019-02-18 by Winter-Guerra