mit-aera/FlightGoggles #44: Starting FlightGoggles fails, NULL exception?

seninp, 2019-02-15:
fresh 16.04 install of ROS kinetic and NVIDIA 410.93... Following the README sequence getting an exception below. Not sure how report and fix this...

```
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=14
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
Vulkan detection: 2
Initialize engine version: 2018.3.0f2 (6e9a27477296)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=14
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: count=55
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance3, enabled=0
[Vulkan init] extensions: name=VK_KHR_multiview, enabled=0
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_relaxed_block_layout, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_sampler_ycbcr_conversion, enabled=0
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_storage_buffer_storage_class, enabled=0
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_variable_pointers, enabled=0
[Vulkan init] extensions: name=VK_EXT_blend_operation_advanced, enabled=0
[Vulkan init] extensions: name=VK_EXT_conservative_rasterization, enabled=0
[Vulkan init] extensions: name=VK_EXT_depth_range_unrestricted, enabled=0
[Vulkan init] extensions: name=VK_EXT_descriptor_indexing, enabled=0
[Vulkan init] extensions: name=VK_EXT_discard_rectangles, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_control, enabled=0
[Vulkan init] extensions: name=VK_EXT_global_priority, enabled=0
[Vulkan init] extensions: name=VK_EXT_post_depth_coverage, enabled=0
[Vulkan init] extensions: name=VK_EXT_sample_locations, enabled=0
[Vulkan init] extensions: name=VK_EXT_sampler_filter_minmax, enabled=0
[Vulkan init] extensions: name=VK_EXT_shader_subgroup_ballot, enabled=0
[Vulkan init] extensions: name=VK_EXT_shader_subgroup_vote, enabled=0
[Vulkan init] extensions: name=VK_EXT_shader_viewport_index_layer, enabled=0
[Vulkan init] extensions: name=VK_EXT_vertex_attribute_divisor, enabled=0
[Vulkan init] extensions: name=VK_NV_clip_space_w_scaling, enabled=0
[Vulkan init] extensions: name=VK_NV_dedicated_allocation, enabled=0
[Vulkan init] extensions: name=VK_NV_device_diagnostic_checkpoints, enabled=0
[Vulkan init] extensions: name=VK_NV_fill_rectangle, enabled=0
[Vulkan init] extensions: name=VK_NV_fragment_coverage_to_color, enabled=0
[Vulkan init] extensions: name=VK_NV_framebuffer_mixed_samples, enabled=0
[Vulkan init] extensions: name=VK_NV_geometry_shader_passthrough, enabled=0
[Vulkan init] extensions: name=VK_NV_sample_mask_override_coverage, enabled=0
[Vulkan init] extensions: name=VK_NV_shader_subgroup_partitioned, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_array2, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_swizzle, enabled=0
[Vulkan init] extensions: name=VK_NVX_device_generated_commands, enabled=0
[Vulkan init] extensions: name=VK_NVX_multiview_per_view_attributes, enabled=0
Vulkan API version 1.1.82
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce GTX 1080 Ti] id=[1b06]
Vulkan device type 2
Vulkan driverversion=[0x66974000] uint=[1721188352]
Vulkan PSO: cache data not found [path=/tmp/MIT-FAST-Lab/FlightGoggles/vulkan_pso_cache.bin]
Begin MonoManager ReloadAssembly
- Completed reload, in  0.060 seconds
WARNING: Shader Unsupported: 'HDRenderPipeline/Lit' - Pass 'META' has no vertex shader
Vulkan: Internal error compiling expression: "u_xlat107 = (_EnableLightLayers != uint(0u)) ? unity_RenderingLayer.x : 3.57331108e-43;
u_xlat107!=0"
 
(Filename:  Line: 65)

Vulkan: Internal error compiling expression: "u_xlat107 = (_EnableLightLayers != uint(0u)) ? unity_RenderingLayer.x : 3.57331108e-43;
u_xlat107!=0"
 
(Filename:  Line: 65)

WARNING: Shader Unsupported: 'Hidden/HDRenderPipeline/TerrainLit_Basemap' - Pass 'META' has no vertex shader
WARNING: Shader Unsupported: 'HDRenderPipeline/TerrainLit' - Pass 'META' has no vertex shader
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 1920 x 1080 @ 60 Hz
InitializeOrResetSwapChain 1024x768 samples=1
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
UnloadTime: 0.709000 ms
Configuring sockets.
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 45)

Trying to connect to: 127.0.0.1
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 45)

Terminated ZMQ sockets.
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 45)

Configuring sockets.
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 45)

Sockets bound.
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 45)

Receiving unhandled NULL exception
#0  0x007f5a6f367ae6 in _nv042glcore
#1  0x00000000963ad6 in std::pow(float, float)
#2  0x000000009a15e3 in std::pow(float, float)
#3  0x00000000965690 in std::pow(float, float)
#4  0x000000009734b1 in std::pow(float, float)
#5  0x00000000918e7d in std::pow(float, float)
#6  0x00000000910d92 in std::pow(float, float)
#7  0x00000000910ec9 in std::pow(float, float)
#8  0x00000000ca3970 in std::pow(float, float)
#9  0x007f5a7584e6ba in start_thread
#10 0x007f5a7558441d in clone
```

varunmurali1, 2019-02-15:
Unfortunately, we have been seeing this issue with some configurations. I believe this is the same issue as #28. Could you replace your unity executable with [this one](https://wiki.archlinux.org/index.php/Vulkan) which has multi threading disabled and check if you still have this issue?

varunmurali1, 2019-02-15:
Sorry, added the wrong link. Meant to add [this link](http://d34kgw45d3q5oc.cloudfront.net/Public/FlightGoggles/StandaloneBinaries/FlightGoggles_Linux_v2.0.0-no_multi_thread.zip)

Winter-Guerra, 2019-02-16:
Is this still an issue for you @seninp?

seninp, 2019-02-16:
In short, yes. The new client fails even faster than the older one. I believe it might be related to the Frankenstein build I put together -- fairly dated mb/cpu/ram combo... SO I was fiddling with that setup and was able to run the regular client few times out of a dozen tries; will investigate further next week. Thank you for following up!

closed 2019-02-16 by seninp