mit-aera/FlightGoggles #45: External rateThrust publisher

ashwinvk94, 2019-02-15:
We remapped the values from a custom joystick and published the values to /uav/inputs/rateThrust.

* The thrust component only seems to take discrete values of ...,8.81,9.81,10.81,...
* The angular rates seem to work only in a binary fashion. 

After that, we tried publishing constant data. The same observations were made about the thrust response. But when we publish even a very low rate of 0.0001 continuously, the drone spins out of control. 

Could you please explain how we should be controlling the drone through a separate publisher?

Note: We are publishing at 20Hz since that was the same rate at which the teleop was publishing.

Winter-Guerra, 2019-02-15:
Hi @ashwinvk94,

It sounds like your method of remapping/publishing input commands has issues. Your binary/stepping behavior sounds similar to that of the keyboard controller, where binary controller is the expected behavior. For testing, we recommend using the Logitech F310 or Xbox One controllers. How are you generating your rateThrust commands? 

The [rateThrust.msg](http://docs.ros.org/kinetic/api/mav_msgs/html/msg/RateThrust.html) accepts arbitrary float values for attitude rates and collective thrust. You are free to publish at any rate you wish, but the inner-loop controller runs at 120Hz in sim-time. 20Hz control sounds very low, but is probably doable. 

Best,
-Winter

Winter-Guerra, 2019-02-16:
@ashwinvk94, are you still having this issue?

ashwinvk94, 2019-02-16:
Hi Winter,

I was busy yesterday. I will send you more information about the problem later today.

ashwinvk94, 2019-02-16:
Hi Winter,

I am attaching two videos which show that the drone responds as expected when the rate inputs are discrete(Video1), but spins out of control initially when the rate inputs are continuous(Video 2). Please not that in video 2 the drone starts to spin out of control only when the thrust set point reaches a certain threshold(approx. 10.81).

The Videos show both the output of the camera and data on `/uav/inputs/rateThrust` through rqt.

[Videos](https://drive.google.com/open?id=1-0ZEnrMvqAdqeNfeOnR1AnQi5wntJ9kW)

Please let me know what you think.

Winter-Guerra, 2019-02-17:
Wow, that's a very odd problem @ashwinvk94. Thanks for the videos! They were extremely helpful.

The takeoff threshold you're seeing makes sense. In version `v2.0.0`, we arbitrarily set this threshold to `10.79N` (`1.1G`) in [this line](https://github.com/mit-fast/FlightGoggles/blob/v2.0.1/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp#L229).  We'll probably change it such that any tiny increase in thrust above `9.81N` (probably `9.82N`) will trigger takeoff in the future. In the meantime, you can change this line yourself to lower the takeoff threshold.

I'm not immediately sure what's going on with the flipping. After the initial flip, you seem to be able to regain control later on in the flight. What commit hash do you currently have checked out for the `flightgoggles` folder? We'll try to recreate the issue from the specific commit you're using. 

Best,
-Winter

koriavinash1, 2019-02-17:
@Winter-Guerra even I'm facing same problem, small RPY rates (like 1e-3) will spin the uav entirely

ashwinvk94, 2019-02-17:
Thanks for the suggestion. I will try lowering the thrust threshold for arming.

The commit hash is : `ebb4d47615f905e3f2b17ef6b0a30c12ecb7204f`

Winter-Guerra, 2019-02-17:
Ok, I think we found the cause of the flipping issue. We're currently testing a fix.

ashwinvk94, 2019-02-17:
Thanks for the help. Hope you find the bug soon.

Winter-Guerra, 2019-02-17:
The new patch is out! Please let us know if this fixes the flipping issue. Additionally, FlightGoggles should reset the drone to the start position after a collision.

```
# Pull new flightgoggles source code
cd ~/catkin_ws/src/flightgoggles
git pull
# Force flightgoggles to redownload binary at build time
catkin clean
catkin build
```

ashwinvk94, 2019-02-17:
Awesome. I'll test it out in a few hours and give you an update.

ashwinvk94, 2019-02-17:
The simulator is working fine now and the quad can be controlled by publishing the values to `/uav/inputs/rateThrust`

closed 2019-02-17 by ashwinvk94