mit-aera/FlightGoggles #46: FlightGoggles Renderer node failes on nvidia-docker

BhavyanshM, 2019-02-15:
I understand that nvidia-docker isn't officially supported. However, I have been trying to setup flightGoggles with all its dependencies inside an nvidia-docker. I have almost everything setup and building just find on the container, but somehow the flightgogglesRenderer node failes when even simply running "rosrun flightgoggles FlightGoggles.x86_64":

Found path: /home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles.x86_64
Mono path[0] = '/home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Managed'
Mono config path = '/home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/MonoBleedingEdge/etc'
Preloaded 'ScreenSelector.so'
Display 0 'U28E590 27"': 3840x2160 (primary device).
Loading player data from /home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/data.unity3d
Logging to /home/docker/.config/unity3d/MIT-FAST-Lab/FlightGoggles/Player.log

I would really appreciate any help at all, and then maybe even contribute my setup in form of a pull request. I am sure such a docker container setup will help out a lot of people that are trying hard to get Flight Goggles setup on their workstations. Thank you in advance!

varunmurali1, 2019-02-15:
I can try and help, but unsure if this is a workable option. Could I have a look at the Player.log?

BhavyanshM, 2019-02-15:
> I can try and help, but unsure if this is a workable option. Could I have a look at the Player.log?

Thank you! Here it is:

docker@58337418fc82:~/catkin_ws$ cat /home/docker/.config/unity3d/MIT-FAST-Lab/FlightGoggles/Player.log
Desktop is 3840 x 2160 @ 60 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 0
No supported renderers found, exiting
 
(Filename:  Line: 634)

varunmurali1, 2019-02-15:
Looks like Vulkan is not able to find a GPU. Could you confirm that you were able to install `apt install -y libvulkan1 mesa-vulkan-drivers vulkan-utils`? Does `nvidia-smi` see a gpu from inside docker? Can you run `vulkan-smoketest` successfully?

BhavyanshM, 2019-02-15:
> Looks like Vulkan is not able to find a GPU. Could you confirm that you were able to install `apt install -y libvulkan1 mesa-vulkan-drivers vulkan-utils`? Does `nvidia-smi` see a gpu from inside docker? Can you run `vulkan-smoketest` successfully?

Yes, I think you are right, the reason is probably that the container isn't able to access the GPU:

docker@58337418fc82:/$ nvidia-smi
Failed to initialize NVML: Driver/library version mismatch
docker@58337418fc82:/$ vulkan-smoketest
terminate called after throwing an instance of 'std::runtime_error'
  what():  VkResult -3 returned
Aborted (core dumped)

Do you see an obvious reason for such a failure to access the GPU from inside the container?

BhavyanshM, 2019-02-15:
> > Looks like Vulkan is not able to find a GPU. Could you confirm that you were able to install `apt install -y libvulkan1 mesa-vulkan-drivers vulkan-utils`? Does `nvidia-smi` see a gpu from inside docker? Can you run `vulkan-smoketest` successfully?
> 
> Yes, I think you are right, the reason is probably that the container isn't able to access the GPU:
> 
> docker@58337418fc82:/$ nvidia-smi
> Failed to initialize NVML: Driver/library version mismatch
> docker@58337418fc82:/$ vulkan-smoketest
> terminate called after throwing an instance of 'std::runtime_error'
> what(): VkResult -3 returned
> Aborted (core dumped)
> 
> Do you see an obvious reason for such a failure in accessing the GPU from inside the container?

varunmurali1, 2019-02-15:
I remember there being a condition with nvidia-docker that the versions of the nvidia driver on the local machine and inside the container should be the same otherwise it throws this error (see [here](https://github.com/NVIDIA/nvidia-docker/wiki/NVIDIA-driver-(version-1.0)))

BhavyanshM, 2019-02-15:
> I remember there being a condition with nvidia-docker that the versions of the nvidia driver on the local machine and inside the container should be the same otherwise it throws this error (see [here](https://github.com/NVIDIA/nvidia-docker/wiki/NVIDIA-driver-(version-1.0)))

Wow, I was just looking at the exact same page. Alright, let me fix this.

BhavyanshM, 2019-02-15:
> Wow, I was just looking at the exact same page. Alright, let me fix this.

So, I am now able to see the ```nvidia-smi``` output detecting my GTX 1070 Ti. However, running flightgoggles crashes with a huge terminal output: 

docker@58337418fc82:/$ rosrun flightgoggles FlightGoggles.x86_64 
No protocol specified
Found path: /home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles.x86_64
Mono path[0] = '/home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Managed'
Mono config path = '/home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/MonoBleedingEdge/etc'
Preloaded 'ScreenSelector.so'
Loading player data from /home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/data.unity3d
Logging to /home/docker/.config/unity3d/MIT-FAST-Lab/FlightGoggles/Player.log
No protocol specified
No protocol specified
No protocol specified
No protocol specified
No protocol specified
No protocol specified
No protocol specified
Stacktrace:

/proc/self/maps:
00400000-01d6c000 r-xp 00000000 08:11 1329922                            /home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles.x86_64
01f6c000-01fe2000 r--p 0196c000 08:11 1329922                            /home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles.x86_64
01fe2000-02015000 rw-p 019e2000 08:11 1329922                            /home/docker/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles.x86_64
02015000-02116000 rw-p 00000000 00:00 0 
03c1a000-04bcf000 rw-p 00000000 00:00 0                                  [heap]
400ae000-4014d000 rw-p 00000000 00:00 0 
40bc8000-40bd8000 rwxp 00000000 00:00 0 
415e0000-415e2000 r-xs 00000000 08:11 1074217                            /tmp/.glybAcKf (deleted)
41d06000-41d16000 rwxp 00000000 00:00 0 
7f74bc000000-7f74bc021000 rw-p 00000000 00:00 0 
7f74bc021000-7f74c0000000 ---p 00000000 00:00 0 
7f74c32a0000-7f74c33a0000 rw-s 00000000 00:06 508                        /dev/nvidiactl
7f74c33a0000-7f74c33a1000 ---p 00000000 00:00 0 
7f74c33a1000-7f74c3ba1000 rw-p 00000000 00:00 0 
7f74c3ba1000-7f74c43a1000 rw-s 00000000 00:06 508                        /dev/nvidiactl
7f74c43a1000-7f74c45a1000 rw-s 00000000 00:06 508                        /dev/nvidiactl
7f74c45a1000-7f74c4ba3000 rw-p 00000000 00:00 0 
7f74c4ba3000-7f74c4fa3000 rw-s 00000000 00:06 508                        /dev/nvidiactl
7f74c4fa3000-7f74c51a3000 rw-s 00000000 00:06 508                        /dev/nvidiactl
7f74c51a3000-7f74c57a5000 rw-p 00000000 00:00 0 
7f74c57a5000-7f74c58a5000 rw-s 00000000 00:06 508                        /dev/nvidiactl
7f74c58a5000-7f74c58e5000 rw-s 00000000 00:06 508                        /dev/nvidiactl
7f74c58e5000-7f74c5905000 rw-s 00000000 00:06 508                        /dev/nvidiactl
7f74c5905000-7f74c5945000 rw-s 00000000 00:06 508                        /dev/nvidiactl
7f74c5945000-7f74c5c71000 r-xp 00000000 08:11 1204607                    /usr/lib/x86_64-linux-gnu/libvulkan_intel.so
7f74c5c71000-7f74c5e70000 ---p 0032c000 08:11 1204607                    /usr/lib/x86_64-linux-gnu/libvulkan_intel.so
7f74c5e70000-7f74c5eb3000 r--p 0032b000 08:11 1204607                    /usr/lib/x86_64-linux-gnu/libvulkan_intel.so
7f74c5eb3000-7f74c5eb4000 rw-p 0036e000 08:11 1204607                    /usr/lib/x86_64-linux-gnu/libvulkan_intel.so
7f74c5eb4000-7f74c5eb7000 rw-p 00000000 00:00 0 
7f74c5eb7000-7f74c5ec5000 r-xp 00000000 08:11 551153                     /usr/lib/x86_64-linux-gnu/libxcb-randr.so.0.1.0
7f74c5ec5000-7f74c60c5000 ---p 0000e000 08:11 551153                     /usr/lib/x86_64-linux-gnu/libxcb-randr.so.0.1.0
7f74c60c5000-7f74c60c6000 r--p 0000e000 08:11 551153                     /usr/lib/x86_64-linux-gnu/libxcb-randr.so.0.1.0
7f74c60c6000-7f74c60c7000 rw-p 0000f000 08:11 551153                     /usr/lib/x86_64-linux-gnu/libxcb-randr.so.0.1.0
7f74c60c7000-7f74c60cf000 r-xp 00000000 08:11 535113                     /usr/lib/x86_64-linux-gnu/libdrm_amdgpu.so.1.0.0
7f74c60cf000-7f74c62ce000 ---p 00008000 08:11 535113                     /usr/lib/x86_64-linux-gnu/libdrm_amdgpu.so.1.0.0
7f74c62ce000-7f74c62cf000 r--p 00007000 08:11 535113                     /usr/lib/x86_64-linux-gnu/libdrm_amdgpu.so.1.0.0
7f74c62cf000-7f74c62d0000 rw-p 00008000 08:11 535113                     /usr/lib/x86_64-linux-gnu/libdrm_amdgpu.so.1.0.0
7f74c62d0000-7f74c62e9000 r-xp 00000000 08:11 425578                     /usr/lib/x86_64-linux-gnu/libelf-0.170.so
7f74c62e9000-7f74c64e8000 ---p 00019000 08:11 425578                     /usr/lib/x86_64-linux-gnu/libelf-0.170.so
7f74c64e8000-7f74c64e9000 r--p 00018000 08:11 425578                     /usr/lib/x86_64-linux-gnu/libelf-0.170.so
7f74c64e9000-7f74c64ea000 rw-p 00019000 08:11 425578                     /usr/lib/x86_64-linux-gnu/libelf-0.170.so
7f74c64ea000-7f74c66fb000 r-xp 00000000 08:11 1204608                    /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so
7f74c66fb000-7f74c68fb000 ---p 00211000 08:11 1204608                    /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so
7f74c68fb000-7f74c6928000 r--p 00211000 08:11 1204608                    /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so
7f74c6928000-7f74c692a000 rw-p 0023e000 08:11 1204608                    /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so
7f74c692a000-7f74c692d000 rw-p 00000000 00:00 0 
[... cut ...]
	/home/docker/catkin_ws/devel/lib/flightgoggles/FlightGoggles.x86_64() [0x4572a9]

Debug info from gdb:

mono_gdb_render_native_backtraces not supported on this platform, unable to find gdb or lldb

=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Aborted (core dumped)

Winter-Guerra, 2019-02-15:
Could you please send the player.log? The terminal output is useless. Does `vulkan-smoketest` pass?

BhavyanshM, 2019-02-15:
> Could you please send the player.log? Does `vulkan-smoketest` pass?

Sure. Here is the player.log:

docker@58337418fc82:/$ cat /home/docker/.config/unity3d/MIT-FAST-Lab/FlightGoggles/Player.log
Desktop is 0 x 0 @ 0 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 2
Initialize engine version: 2018.3.0f2 (6e9a27477296)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: count=46
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_relaxed_block_layout, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_sampler_ycbcr_conversion, enabled=0
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_storage_buffer_storage_class, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_variable_pointers, enabled=0
[Vulkan init] extensions: name=VK_KHX_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHX_multiview, enabled=0
[Vulkan init] extensions: name=VK_EXT_blend_operation_advanced, enabled=0
[Vulkan init] extensions: name=VK_EXT_depth_range_unrestricted, enabled=0
[... 20 lines cut ...]
Vulkan API version 1.0.65
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce GTX 1070 Ti] id=[1b82]
Vulkan device type 2
Vulkan driverversion=[0x6195c000] uint=[1637203968]
Vulkan PSO: cache data not found [path=/tmp/MIT-FAST-Lab/FlightGoggles/vulkan_pso_cache.bin]
Receiving unhandled NULL exception
#0  0x0000000097bd08 in std::pow(float, float)
#1  0x0000000097c00e in std::pow(float, float)
#2  0x0000000090266f in std::pow(float, float)
#3  0x00000000cd5842 in std::pow(float, float)
#4  0x00000000cd8e42 in std::pow(float, float)
#5  0x00000000c2dde5 in std::pow(float, float)
#6  0x00000000c40883 in std::pow(float, float)
#7  0x0000000044a54e in (Unknown)
#8  0x007f75712edb97 in __libc_start_main
#9  0x000000004572a9 in (Unknown)

The ```vulkan-smoketest``` output:

docker@58337418fc82:/$ vulkan-smoketest
No protocol specified
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to connect to the display server
Aborted (core dumped)

Winter-Guerra, 2019-02-15:
OK, there seem to be two different errors here. Vulkan-smoketest is not picking up your display. Perhaps you need to do some X tunneling/forwarding? You also should run the [single-threaded binary for FlightGoggles](http://d34kgw45d3q5oc.cloudfront.net/Public/FlightGoggles/StandaloneBinaries/FlightGoggles_Linux_v2.0.0-no_multi_thread.zip).

BhavyanshM, 2019-02-15:
> OK, there seem to be two different errors here. Vulkan-smoketest is not picking up your display. Perhaps you need to do some X tunneling/forwarding? You also should run the [single-threaded binary [...]

BhavyanshM, 2019-02-15:
> OK, there seem to be two different errors here. Vulkan-smoketest is not picking up your display. Perhaps you need to do some X tunneling/forwarding? You also should run the [single-threaded binary [...]

Ahh, didn't mean to close the issue. Just hit the "Close and comment" button unintentionally. 

The ```vulkan-smoketest``` seems to launch a window and then exit with a segmentation fault error. I should also mention that the first time I launched mission planner, the display monitor basically just crashed and I had to restart the system. I do have X tunneling configured on the docker container according to this page: http://wiki.ros.org/docker/Tutorials/Hardware%20Acceleration. Working on launching the single threaded version.

Winter-Guerra, 2019-02-15:
No problem! In that case, I would advise getting `glxgears` to run first, then `vulkan-smoketest`, then Flightgoggles single-threaded version.

BhavyanshM, 2019-02-15:
> glxgears

```glxgears``` works now after installing ```mesa-utils``` but ```vulkan-smoketest``` still fails with a segmentation fault.

Winter-Guerra, 2019-02-15:
OK, what about the output of `vulkaninfo`? Could you please post that here?

Do you also have vulkan installed in the host OS?

BhavyanshM, 2019-02-15:
> OK, what about the output of `vulkaninfo`? Could you please post that here?
> 
> Do you also have vulkan installed in the host OS?

Yes, Vulkan is installed on the host OS as well. However, here is the output from inside the container for ```vulkaninfo```

```docker@58337418fc82:/$ vulkan-smoketest
Segmentation fault (core dumped)
docker@58337418fc82:/$ rviz
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-docker'
[ INFO] [1550206886.510225422]: rviz version 1.13.1
[ INFO] [1550206886.510273992]: compiled against Qt version 5.9.5
[ INFO] [1550206886.510285611]: compiled against OGRE version 1.9.0 (Ghadamon)
docker@58337418fc82:/$ vulkan-smoketest --verbose
Segmentation fault (core dumped)
docker@58337418fc82:/$ vulkan-smoketest --help
Segmentation fault (core dumped)
docker@58337418fc82:/$ vulkaninfo
===========
VULKAN INFO
===========

Vulkan Instance Version: 1.1.70

Instance Extensions:
====================
Instance Extensions	count = 16
	VK_EXT_acquire_xlib_display         : extension revision  1
	VK_EXT_debug_report                 : extension revision  9
	VK_EXT_direct_mode_display          : extension revision  1
	VK_EXT_display_surface_counter      : extension revision  1
	VK_KHR_display                      : extension revision 21
	VK_KHR_get_physical_device_properties2: extension revision  1
	VK_KHR_get_surface_capabilities2    : extension revision  1
	VK_KHR_surface                      : extension revision 25
	VK_KHR_xcb_surface                  : extension revision  6
	VK_KHR_xlib_surface                 : extension revision  6
	VK_KHR_external_fence_capabilities  : extension revision  1
	VK_KHR_external_memory_capabilities : extension revision  1
	VK_KHR_external_semaphore_capabilities: extension revision  1
	VK_KHR_device_group_creation        : extension revision  1
	VK_KHR_wayland_surface              : extension revision  6
	VK_EXT_debug_utils                  : extension revision  1
Layers: count = 0
=======
Presentable Surfaces:
=====================
GPU id       : 0 (GeForce GTX 1070 Ti)
Surface type : VK_KHR_xcb_surface
Formats:		count = 0
Present Modes:		count = 0

VkSurfaceCapabilitiesKHR:
=========================
	minImageCount       = 2
	maxImageCount       = 8
	currentExtent:
		width       = 256
		height      = 256
	minImageExtent:
		width       = 256
		height      = 256
	maxImageExtent:
		width       = 256
		height      = 256
	maxImageArrayLayers = 1
	supportedTransform:
		VK_SURFACE_TRANSFORM_IDENTITY_BIT_KHR
	currentTransform:
		VK_SURFACE_TRANSFORM_IDENTITY_BIT_KHR
[... 2704 lines cut ...]
	linearTiling   FormatFeatureFlags:
		None
	optimalTiling  FormatFeatureFlags:
		None
	bufferFeatures FormatFeatureFlags:
		None

FORMAT_ASTC_12x12_UNORM_BLOCK:
	linearTiling   FormatFeatureFlags:
		None
	optimalTiling  FormatFeatureFlags:
		None
	bufferFeatures FormatFeatureFlags:
		None

FORMAT_ASTC_12x12_SRGB_BLOCK:
	linearTiling   FormatFeatureFlags:
		None
	optimalTiling  FormatFeatureFlags:
		None
	bufferFeatures FormatFeatureFlags:
		None
```

Also, here are the files in case you need to look at them. https://github.com/BhavyanshM/ROSEnv/tree/master/Melodic

BhavyanshM, 2019-02-15:
On simply running: ```rosrun flightgoggles FlightGoggles.x86_64```

Container terminal outputs the following and crashes the monitor display.

[image]

Here is the output on Player.log from this attempt to launch flightgoggles:

```Desktop is 3840 x 2160 @ 60 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 2
Initialize engine version: 2018.3.0f2 (6e9a27477296)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: count=46
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_relaxed_block_layout, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_sampler_ycbcr_conversion, enabled=0
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_storage_buffer_storage_class, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_variable_pointers, enabled=0
[Vulkan init] extensions: name=VK_KHX_device_group, enabled=0
[... 36 lines cut ...]

FMOD failed to initialize the output device.: "Error initializing output device. " (60)
 
(Filename:  Line: 1398)

FMOD failed to initialize any audio devices, running on emulated software output with no sound. Please check your audio drivers and/or hardware for malfunction.
Begin MonoManager ReloadAssembly
- Completed reload, in  0.077 seconds
WARNING: Shader Unsupported: 'HDRenderPipeline/Lit' - Pass 'META' has no vertex shader
Vulkan: Internal error compiling expression: "u_xlat107 = (_EnableLightLayers != uint(0u)) ? unity_RenderingLayer.x : 3.57331108e-43;
u_xlat107!=0"
 
(Filename:  Line: 65)

Vulkan: Internal error compiling expression: "u_xlat107 = (_EnableLightLayers != uint(0u)) ? unity_RenderingLayer.x : 3.57331108e-43;
u_xlat107!=0"
 
(Filename:  Line: 65)

WARNING: Shader Unsupported: 'Hidden/HDRenderPipeline/TerrainLit_Basemap' - Pass 'META' has no vertex shader
WARNING: Shader Unsupported: 'HDRenderPipeline/TerrainLit' - Pass 'META' has no vertex shader
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 3840 x 2160 @ 60 Hz
InitializeOrResetSwapChain 1024x768 samples=1```

varunmurali1, 2019-02-15:
It looks like Unity3d is attempting to access the audio driver, one option is getting pulseaudio to work in docker. We will also try to build you a binary with sound disabled to see if that helps mitigate this issue.

BhavyanshM, 2019-02-15:
> It looks like Unity3d is attempting to access the audio driver, one option is getting pulseaudio to work in docker. We will also try to build you a binary with sound disabled to see if that helps mitigate this issue.

Looks like it is non-trivial to get pulseaudio to work in docker. So, I would really appreciate a binary with sound disabled.

varunmurali1, 2019-02-15:
Sure, we will try to generate a binary with sound disabled. In the meantime would it be possible to share the Dockerfile so I can also take a quick look at it?

BhavyanshM, 2019-02-15:
> Sure, we will try to generate a binary with sound disabled. In the meantime would it be possible to share the Dockerfile so I can also take a quick look at it?

Oh sure, I had actually posted it at the end of an earlier comment but here are all the files. https://github.com/BhavyanshM/ROSEnv/tree/master/Melodic

Winter-Guerra, 2019-02-15:
We've just updated the binary. This build should not try to connect to the audio card. Please let us know if it works! @BhavyanshM

BhavyanshM, 2019-02-16:
> We've just updated the binary. This build should not try to connect to the audio card. Please let us know if it works! @BhavyanshM

Thank you @Winter-Guerra. So I tried to ```catkin clean flightgoggles``` and then ```catkin build```, however this time, the download only took a minute, which was somewhat unusual considering the earlier 10 min time. Assuming that the new binary got successfully downloaded, I tried to launch ```rosrun flightgoggles FlightGoggles.x86_64``` again. However, I was met with the same problem of my monitoring crashing with the following terminal output.

[image]

Also, the ```vulkan-smoketest``` outputs the following:
```docker@58337418fc82:~/catkin_ws$ vulkan-smoketest
No protocol specified
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to connect to the display server
Aborted (core dumped)```

Winter-Guerra, 2019-02-16:
Did you pull the new changes to the repo before rebuilding? 1 min download
sounds like catkin was not updated and cached your previous version.
```
wstool update
catkin clean
catkin build
```

BhavyanshM, 2019-02-16:
> Did you pull the new changes to the repo before rebuilding? 1 min download sounds like catkin was not updated and cached your previous version. ``` cd ~/catkin_ws/src && wstool update && catkin clean && catkin build ```

So no matter how many times I perform the steps with ```wstool update``` with ```catkin clean``` and ```catkin build```, the flightgoggles module always ends in just about 10 seconds. I even tried manually deleting the build, source and logs directories but still it builds back in just a few seconds again. It takes about a min on the host, but on the container the build just ends in 30-40 seconds. Anyhow, I still tried all your commands and tried to launch again. Anyhow, here is the player.log from this run.

```docker@58337418fc82:/$ cat /home/docker/.config/unity3d/MIT-FAST-Lab/FlightGoggles/Player.log
Desktop is 3840 x 2160 @ 60 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 2
Initialize engine version: 2018.3.0f2 (6e9a27477296)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: count=46
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_relaxed_block_layout, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_sampler_ycbcr_conversion, enabled=0
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_storage_buffer_storage_class, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_variable_pointers, enabled=0
[Vulkan init] extensions: name=VK_KHX_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHX_multiview, enabled=0
[Vulkan init] extensions: name=VK_EXT_blend_operation_advanced, enabled=0
[Vulkan init] extensions: name=VK_EXT_depth_range_unrestricted, enabled=0
[... 12 lines cut ...]
[Vulkan init] extensions: name=VK_NV_glsl_shader, enabled=0
[Vulkan init] extensions: name=VK_NV_clip_space_w_scaling, enabled=0
[Vulkan init] extensions: name=VK_NV_sample_mask_override_coverage, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_array2, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_swizzle, enabled=0
[Vulkan init] extensions: name=VK_NV_geometry_shader_passthrough, enabled=0
[... cut ...]
- Completed reload, in  0.048 seconds
WARNING: Shader Unsupported: 'HDRenderPipeline/Lit' - Pass 'META' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/HDRenderPipeline/TerrainLit_Basemap' - Pass 'META' has no vertex shader
WARNING: Shader Unsupported: 'HDRenderPipeline/TerrainLit' - Pass 'META' has no vertex shader
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 3840 x 2160 @ 60 Hz
InitializeOrResetSwapChain 1024x768 samples=1```

```vulkan-smoketest``` still fails with a segmentation fault though.

Winter-Guerra, 2019-02-17:
Hi @BhavyanshM,

OK, so it looks like the new `v2.0.1` binary fixed the pulseaudio problem. However, Vulkan still crashes inside of the `nvidia-docker` image. 

Unfortunately, it looks like `nvidia-docker` [does not support the Vulkan renderer yet](https://github.com/NVIDIA/nvidia-docker/wiki/Frequently-Asked-Questions#is-vulkan-supported). Here's the [corresponding issue link](https://github.com/NVIDIA/nvidia-docker/issues/631). Until they add support in `nvidia-docker` for the Vulkan render engine, I don't think that there is any way to render the new `v2.0.x` environments from within Docker. 

FlightGoggles `v1.8.0` uses the OpenCL/OpenGL rendering backend, but does not support the new environments used in the AlphaPilot challenge. 

I'm terribly sorry for the inconvenience @BhavyanshM. At the moment, it looks like it is not possible to run FlightGoggles `v2.0.x` on `nvidia-docker` due to lack of support for the Vulkan renderer in `nvidia-docker`. Older versions of FlightGoggles should run on `nvidia-docker`, but do not support the new environment used in the AlphaPilot challenge. 

If you'd really like to run the FlightGoggles simulation inside of Docker, you could run the FlightGoggles renderer binary outside of Docker in Linux, Windows, or MacOS and point the client IP at the ROS docker image. Currently, we only have the [Linux binary published for `v2.0.1`](https://github.com/mit-fast/FlightGoggles/releases/tag/v2.0.1) and Windows and Linux published for [`v2.0.0`](https://github.com/mit-fast/FlightGoggles/releases/tag/v2.0.0) but will add Windows and MacOS binaries of `v2.0.1` by sometime Monday. 

Best,
-Winter

Winter-Guerra, 2019-02-18:
Closing due to lack of `nvidia-docker` support for Vulkan.

closed 2019-02-18 by Winter-Guerra

BhavyanshM, 2019-02-18:
> Hi @BhavyanshM,
>
> OK, so it looks like the new `v2.0.1` binary fixed the pulseaudio problem. However, Vulkan still crashes inside of the `nvidia-docker` image.
>
> Unfortunately, it looks like [...]

No worries @Winter-Guerra It is unfortunate that ```nvidia-docker``` does not yet support vulkan. However, the alternative of running the FG binary outside the container is quite a practical solution. The containerization of the ROS packages required still takes away quite a lot of work from the end-users. Thank you for all the support anyway!

edowson, 2019-04-14:
See #131 for a solution. You'll have to script these into a Dockerfile.

01. Creating file /usr/share/vulkan/icd.d/nvidia_icd.json with the following contents:

```json
{
    "file_format_version" : "1.0.0",
    "ICD": {
        "library_path": "libGLX_nvidia.so.0",
        "api_version" : "1.1.95"
    }
}
```

02. Download the nvidia drivers, extract the *.run package and manually extract the libnvidia-glvkspirv.so.418.56 file to /usr/lib/x86_64-linux-gnu/, inside the docker image.

This will allow vulcaninfo and vulkan-smoketest to run correctly.

Running `rosrun flightgoggles FlightGoggles.x86_64` pops up the render window.
```bash
wget http://us.download.nvidia.com/XFree86/Linux-x86_64/418.56/NVIDIA-Linux-x86_64-418.56.run
NVIDIA-Linux-x86_64-418.56.run --extract-only
sudo cp NVIDIA-Linux-x86_64-418.56/libnvidia-glvkspirv.so.418.56 /usr/lib/x86_64-linux-gnu/
```