mit-aera/FlightGoggles #47: Running Simulator on TX2

Kazimbalti, 2019-02-15:
I am not able to run the FlightGoggles sim on Nvidia TX2. Please let me know that we can use TX2 for this simulator? because when I run it on TX2 it gives me an error. Thank you

ashwinvk94, 2019-02-15:
I think it would be better if you post the details of the error which you encountered, so that others can help debug it.

Kazimbalti, 2019-02-15:
I have this  problem when running roslaunch flightgoggles teleopExample.launch

Unable to launch [uav/flightgogglesRenderer-6]. 
If it is a script, you may be missing a '#!' declaration at the top.
The traceback for the exception was written to the log file

Kazimbalti, 2019-02-15:
when I will run with roslaunch flightgoggles teleopExample.launch use_external_renderer:=1

The RViz will open and shows No Image on RViz

Winter-Guerra, 2019-02-15:
Running the renderer on a TX2 is not supported or possible. It is possible to run the renderer on a different computer and stream the images across Ethernet to the TX2. This can be done by running the binary on  a different computer running Linux/Mac/Windows and changing the FloghtGoggles renderer IP address on the splash screen to the TX2 client.

closed 2019-02-15 by Winter-Guerra

neotonight, 2019-03-21:
i try to run at tx2 
but 
same problem 
[image]

Winter-Guerra, 2019-03-21:
Hi everyone,

We've created [a wiki page regarding how to run hardware-in-the-loop simulations using FlightGoggles and the NVIDIA Jetson TX2](https://github.com/mit-fast/FlightGoggles/wiki/Connecting-FlightGoggles-to-an-embedded-system). Hopefully this wiki page has answers to some of your questions! Please open another GitHub issue if you're still having trouble after following the directions on the wiki page.

Thanks,
-Winter