mit-aera/FlightGoggles #48: Renderer crashing

SimpleUAVs, 2019-02-16:
The Unity renderer crashes after a few minutes. Logs attached.
 System is Dell t3600, 16GB ram, AMD 480 GPU.

[crashlog1.txt](https://github.com/mit-fast/FlightGoggles/files/2871700/crashlog1.txt)

[uav-flightgogglesRenderer-5-stdout.log.txt](https://github.com/mit-fast/FlightGoggles/files/2871701/uav-flightgogglesRenderer-5-stdout.log.txt)
[Player.log.txt](https://github.com/mit-fast/FlightGoggles/files/2871702/Player.log.txt)

Winter-Guerra, 2019-02-16:
What version number do you see on the renderer splash screen?

Winter-Guerra, 2019-02-18:
@SimpleUAVs do you still have this issue using the new renderer version?

```bash
# Pull new flightgoggles source code
cd ~/catkin_ws/src/flightgoggles
git pull
# Force flightgoggles to redownload binary at build time
catkin clean
catkin build
```

SimpleUAVs, 2019-02-18:
Did this today, Isn’t crashing anymore, but still cannot get joystick/rc controller to work. Working on it.

 

jz

Winter-Guerra, 2019-02-18:
Closing issue since this issue (renderer crashing) has been resolved.

closed 2019-02-18 by Winter-Guerra