mit-aera/FlightGoggles #50: ver.:2.0.2: again multithreading?

lukked, 2019-02-17:
I'm a little bit confused because I thought 2.0.1 changed to single threaded.
https://github.com/mit-fast/FlightGoggles/commit/bf53fb7453aac5b472e8087cf17b3eccbcf00a65

I'm having the "unhandled NULL exception" issue, so I tried to use the 2.0.0 no multithread renderer with FG 2.0.2. But now clock scaling doesn't seem to work and the rviz 3d display is unsynchronized with the rendered image.

2.0.2 with 2.0.0 no multithread renderer:
rostopic hz /tf
average rate: 960.000

2.0.0 with 2.0.0 no multithread renderer:
average rate: 120.000

Winter-Guerra, 2019-02-18:
Could you please post your player.log for `v2.0.2`? The renderer should not be using the experimental render pipeline (it was built without it). This might be a different issue.

lukked, 2019-02-18:
This is the unity player.log
[Player.log](https://github.com/mit-fast/FlightGoggles/files/2875012/Player.log)
Error messages from rosrun flightgoggles FlightGoggles.x86_64
[stderr.txt](https://github.com/mit-fast/FlightGoggles/files/2875013/stderr.txt)
i also received some cmake warnings (as far as I can remember the same ones like when building the 2.0.0 multithread version).
[build_warnings.txt](https://github.com/mit-fast/FlightGoggles/files/2875014/build_warnings.txt)

Winter-Guerra, 2019-02-18:
It looks like the Unity backend cannot load some library files. This might be from an incomplete download. I suspect that error is the core issue, not multithreading.

Could you please try downloading v2.0.2 from the releases page and try again? Alternatively, `catkin clean && catkin build`

```
Fallback handler could not load library /home/drg/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32
Fallback handler could not load library /home/drg/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32.so
Fallback handler could not load library /home/drg/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/Kernel32
Fallback handler could not load library /home/drg/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32
Fallback handler could not load library /home/drg/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32.so
Fallback handler could not load library /home/drg/catkin_ws/devel/.private/flightgoggles/lib/flightgoggles/FlightGoggles_Data/Mono/libKernel32
```

lukked, 2019-02-18:
I did it allready. These missing library messages appeared already in the first release, and then with 2.0.0_no_multithread everything was fine.

lukked, 2019-02-18:
Here is the log from FG 2.0.2 with renderer 2.0.0_no_multithread copied into the devel/lib directory
It works but out of sync as I already mentioned.
[Player.log](https://github.com/mit-fast/FlightGoggles/files/2876775/Player.log)

Winter-Guerra, 2019-02-18:
Hmm, ok. That's odd. Could you please tell me a bit more about your
system/OS/Drivers?

lukked, 2019-02-18:
ubuntu 16.04 last upgrade. nvidia 384.130. vulkan-smoketest works at 60FPS. 
Its a i5 laptop with an internal radeon and an external nvidia 1050 ti. monitors attached to nvidia. and graphics and opengl go through that.

Winter-Guerra, 2019-02-18:
Ok, this is probably due to binary build using Unity3D backend `2018.3.0f2`
vs `2018.3.5x`. Unity3D changelog mentions some performance and stability
fixes for Vulkan. I'll try building a binary for you with a newer backend.

lukked, 2019-02-18:
OK its kubuntu not ubuntu but i think that shouldn't matter. i' just more used to having kde.

now i see that in the linked stderr.txt  /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so is mentioned besides nvidia.
That's the only weird thing I can see right now.
I don't know why because /usr/lib/xorg/Xorg certainly using nvidia (it is listed in nvidia-smi)

I'm unable to unload radeon driver although removed it from the pci bus so it's not visible with lspci and the radeno GPU is not usable by the system.

lukked, 2019-02-18:
> Ok, this is probably due to binary build using Unity3D backend `2018.3.0f2` vs `2018.3.5x`. Unity3D changelog mentions some performance and stability fixes for Vulkan. I'll try building a binary for you with a newer backend.
> […](#)

OK, thanks a lot.

lukked, 2019-02-18:
Just in case that I was not completely clear about the issues I am experiencing:

FG 2.0.0 with renderer 2.0.0_no_multithread actually works very well on my system. The graphics is rather slow (I know that the PCI connection is bottlenecking the eGPU) but ROS clock scaling does its job and the drone is controllable by the keyboard. 

This is not the case with FG 2.0.2 and the same renderer where the clock scaling just minimally slows down ROS, not enough to make the drone controllable.

The original (experimental render pipeline) 2.0.0 and the newest (no experimental render pipeline) 2.0.2 renderers both crash with, oddly, the same error messages.

lukked, 2019-02-19:
v2.0.3 doesn't crash anymore!
Thank you!

Regarding the lack of synchronization on slow hardware setups: this appears to be a separate issue. I checked older versions and this behavior started after this https://github.com/mit-fast/FlightGoggles/commit/bf53fb7453aac5b472e8087cf17b3eccbcf00a65. 
It seems that clock scaling can be set manually. Could you please tell me how?

Winter-Guerra, 2019-02-19:
Manual clock scaling is done by setting clockscale in https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles/config/drone/drone.yaml

For automatic 60hz clock scale, this line should be commented out (#).

Thanks for the heads up. We found a bug in the code that allows for clock scale when using non-sim time (which should not be possible) that will be patched later today.

lukked, 2019-02-19:
Great, manual scaling works. I'm fine for now. Thanks again.

closed 2019-02-19 by Winter-Guerra

PHRABAL, 2019-02-19:
@Winter-Guerra I'm getting confused by the changes in some of these items and settings related to sim time, clock speed, etc. from v2.0.0 to current. I noticed with v2.0.2 on startup it says the use_sim_time parameter is not set and so using the default false. Above you talk about manual clock scaling. Just a basic question - to run my autonomous algo on FG, with the p3, what should my settings be for sim_time, clock speed, etc.? Are the defaults in v2.0.2 correct or do I need to change something?

Winter-Guerra, 2019-02-19:
Just pushed a patch for automatic clock scaling. However, if manual works for you then I guess it doesn’t really matter @lukked . I’m glad we were able to solve your original problems!

@PHRABAL when LM evaluates your algorithm, I think that they will set `/use_sim_time` and unset `/uav/flightgoggles_uav_dynamics/clockscale` to enable automatic clock scaling. The latest commit should have been a bugfix and should not have really changed the behavior of the clock scaling system.

If you’d like to use manual clock scaling, set `/use_sim_time` and set `/uav/flightgoggles_uav_dynamics/clockscale` to your desired clock scaling.

Winter-Guerra, 2019-02-19:
@PHRABAL we now have a lot of code and params documentation and FAQ answers on the wiki: 
https://github.com/mit-fast/FlightGoggles/wiki

PHRABAL, 2019-02-19:
@Winter-Guerra thank you. So when I launch core in v2.0.2, it says `Did not get bool useSimTime_ from the params, defaulting to false`. So what you are suggesting is that your default here (not use sim time) is not what we will want to use for the competition. We should 1) enable sim time and 2) comment out clock scaling.

Winter-Guerra, 2019-02-19:
I actually suggest that you upgrade to v2.0.3 (latest commit on master)
@PHRABAL. I think that the latest commit should make core.launch use
automatic clock scaling.

PHRABAL, 2019-02-19:
@Winter-Guerra upgraded to v2.0.3. `core.launch` now uses auto clock scaling. But still see `Did not get bool useSimTime_ from the params, defaulting to false` upon launch, so have added `<param name="/use_sim_time" value="false" />` to bottom of `core.launch`.

Winter-Guerra, 2019-02-19:
That's odd. Core.launch should set the use_sim_time parameter to true (
https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles/launch/core.launch#L14).
Additionally, auto scaling cannot be enabled if `use_sim_time` is false.
Will investigate more.

PHRABAL, 2019-02-19:
https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles/launch/core.launch#L14 is not in my core.launch file in v2.0.3.

Once I set sim time to true, at launch it correctly says `Using sim_time and did not get a clock scaling value. Defaulting to automatic clock scaling.`

Not sure why your line in core.launch didn't make it into v2.0.3.

Winter-Guerra, 2019-02-19:
Could you please pull again from master? It sounds like your local branch
is a few commits behind master.

PHRABAL, 2019-02-19:
@Winter-Guerra This might help you: When I upgraded from v2.0.0 to v.2.0.2 on Sunday it first kicked off an error saying that a couple files it wanted to update had been modified by me and I had to remove those files before the update would put the replace files there and complete the update. But today going from v2.0.2 to v.2.0.3 I did not get that error even though I had manually modified a couple files that should have been updated with v.2.0.3, like `drone.yaml`.

PHRABAL, 2019-02-19:
@Winter-Guerra Never mind that comment. It did kick the error. The core.launch didn't upgrade because I missed the error message as it scrolled off the screen.

commit b1202a9f (2019-02-19): Fixed bug where clockscaling could be enabled when not using sim time.

Also makes automatic clock scaling while using sim time the default option. Closes #50