mit-aera/FlightGoggles #52: Height sensor support

ashwinvk94, 2019-02-18:
Is there any way of getting drone height feedback data through the simulator?

Winter-Guerra, 2019-02-18:
Hi @ashwinvk94,
This sensor will be added to the simulator sometime Monday. It will act as a single-point downward facing lidar. We were not informed that this sensor was required by Lockheed for the simulator until after Friday's release. Stay tuned!

Winter-Guerra, 2019-02-19:
Hi @ashwinvk94,

We've just pushed a patch for `v2.0.3`. It adds a downward facing laser range finder for height sensing. To update, please run the following commands.
 
```bash
# Pull new flightgoggles source code
cd ~/catkin_ws/src/flightgoggles
git pull
# Force flightgoggles to redownload binary at build time
catkin clean
catkin build
```

Best,
-Winter

closed 2019-02-19 by Winter-Guerra