mit-aera/FlightGoggles #53: Catkin build and FG binary download should be decoupled if possible.

BhavyanshM, 2019-02-18:
Hello,

This issue concerns with the step in the catkin build process which involves the download of the binary. Quite often it is hard to discern the reason behind long build times for the ```flightgoggles``` catkin package, since it downloads the FG binary from the HTTP/FTP servers in the background. It would be great if this step could be separated out of the catkin build step. Sometimes the entire build fails if catkin can simply not connect to the servers. The user should be allowed to separately download the binary from the releases page and place it in the appropriate directory. It will basically increase transparency into the whole build process and offer flexibility in updating flightgoggles.

varunmurali1, 2019-02-19:
Thank you for the suggestion. Although I agree with you that decoupling the process is probably a good idea, we will not be adding this feature at this time. 

I am leaving the issue open as a feature request. If our constraints allow the addition and testing of this feature, we will implement this feature in the future.

Winter-Guerra, 2019-03-20:
Added ability to turn off automatic download of FlightGoggles binary in Catkin via `-DFLIGHTGOGGLES_DOWNLOAD_BINARY=ON/OFF`. Closing.

closed 2019-03-20 by Winter-Guerra