mit-aera/FlightGoggles #56: IMU sensor topic publish start time

juanmed, 2019-02-18:
Hello,

I see the topic /uav/sensors/imu is being published only after messages to /uav/inpu/rateThrust start arriving. Is this the normal behaviour? Is there any reason for this?

eatal, 2019-02-18:
Hi @juanmed,

The reason is that currently the dynamics simulation only starts once a thrust input of at least 1.1 times the vehicle weight (1.1*9.81) is received. This is to prevent the drone from immediately drifting away from its initial position (mainly due to IMU noise), once FlightGoggles is started.

The IMU messages are only published if the dynamics simulation is running, hence those only start arriving after the initial thrust command.

-Ezra

Winter-Guerra, 2019-02-19:
Hi @juanmed,

FYI, we've lowered the takeoff threshold to 9.9N. After inputting a thrust command `>9.9N`, the drone should takeoff and start sending IMU messages.

Best,
-Winter

closed 2019-02-19 by Winter-Guerra