mit-aera/FlightGoggles #58: Unity crashes on latest master

jayavanth, 2019-02-20:
When I run any of the executables that launch Unity, I get these errors and Unity crashes:

[trajPlayback.launch](https://pastebin.com/4S9NNmve)

[Player.log](https://pastebin.com/a7Gtx3sd)

I tried using the binary in #28 and Unity still crashes

OS: Ubuntu 16.04
NV driver: 410.79
Unity: 7.4.5

Winter-Guerra, 2019-02-20:
Hi @jayavanth ,

Could you please try running `vulkan-smoketest`? Does that run? It is a utility from `sudo apt install vulkan-utils`.

Best,
-Winter

jayavanth, 2019-02-20:
@Winter-Guerra `vulkan-smoketest` runs fine

varunmurali1, 2019-02-20:
@jayavanth We are still trying to figure this out. Can you tell me how much VRAM is available on the GPU? I can see that you are using a GT 730 and it is likely you are running out of VRAM. Can you please post the output of `nvidia-smi` without running any of FlightGoggles?

jayavanth, 2019-02-20:
@varunmurali1 it's just 2GB and ~200MB is already allocated before launching FlightGoggles. I'll swap it out with a better card and see if things change. How much VRAM does FlightGoggles need?

Winter-Guerra, 2019-02-21:
Hi @jayavanth ,

Unfortunately, your GPU’s VRAM size is probably the cause of this issue. From the FAQ page on the wiki:
> Q. What are the minimum requirements to run FlightGoggles locally?
> At the moment, FlightGoggles requires ~2.4GB of VRAM and has the same underlying requirements as the Vulkan rendering API.

I’m closing this issue for the moment since this setup falls outside of the supported set of specifications that FlightGoggles supports. However, please feel free to reopen this issue if you continue to have the problem after upgrading your GPU.

Best of luck!
-Winter

closed 2019-02-21 by Winter-Guerra

jayavanth, 2019-02-21:
Thanks! I switched out the GPU to a Titan X with 12G but now I'm getting some other errors. I can see a glimpse of the simulation on Unity and then it gets back to disconnected state. Logs attached

[roslaunch flightgoggles reporter.launch level:=easy] (https://pastebin.com/82f4Whhx)
[Player.log](https://pastebin.com/KbHmq2mu)

varunmurali1, 2019-02-21:
Please open this as a new issue, because this is unrelated to the original issue.

Winter-Guerra, 2019-02-21:
Hi @jayavanth ,

This issue is caused by a version mismatch between the ROS code and the downloaded FG binary. Please follow the instructions here to update both to the latest version: 
https://github.com/mit-fast/FlightGoggles/wiki/Updating-to-Latest-Version

Best,
-Winter

jayavanth, 2019-02-21:
Thanks, it worked!