mit-aera/FlightGoggles #59: New v2.0.3 error: Could not import "pyqt" bindings

PHRABAL, 2019-02-20:
Hi, after installing v2.0.3 and running core.launch, I get this error below. FG seems to work just fine anyway. Is it a problem? I was able to reproduce this issue on 3 different p3 servers that all were upgraded from v2.0.2 to v2.0.3 following your instructions. 

The first time core launches after upgrading it takes a couple minutes to output the error, but then every launch thereafter the error appears immediately.

```
Could not import "pyqt" bindings of qt_gui_cpp library - so C++ plugins will not be available:
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ImportError: dynamic module does not define module export function (PyInit_libqt_gui_cpp_sip)
```

Winter-Guerra, 2019-02-20:
I think that this comes from a convenience plotting node I added to the core.launch file for plotting the laser range finder readings. You can disable this node in the launch file (https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles/launch/core.launch#L52). It's not necessary, just added for convenience. I've disabled this in master since it is not necessary for core operation of FlightGoggles.

Winter-Guerra, 2019-02-20:
Closing issue since I think that the latest patch fixed this issue. Please reopen if needed.

closed 2019-02-20 by Winter-Guerra