mit-aera/FlightGoggles #60: Feature_Request: ROS paremeter and changes to set camera angle

SimpleUAVs, 2019-02-20:
Camera angle has a great effect on how fast a racing drone can fly any course. Most are set at 30deg up or more. I've even seen 80deg. The FlightGoggles sim is locked to 0deg camera angle. This angle should be adjustable and set by each team individually.

eatal, 2019-02-20:
Hi @SimpleUAVs,
This is a great point. The camera orientation can be changed by changing the appropriate transforms. We will make this feature more easily accessible soon. However, we are not sure if changing the camera orientation is allowed in the challenge. That would be a question for the AlphaPilot organization. Maybe you can raise this on the HeroX forum?

Hope this helps
-Ezra

closed 2019-02-20 by eatal

SimpleUAVs, 2019-02-20:
Will do.