mit-aera/FlightGoggles #61: Feature_Request: make start throttle a ROS parameter and configurable

SimpleUAVs, 2019-02-20:
The base throttle (vertical) value is currently 9.81. This should be set to 0 and/or be configurable via a ROS parameter. At 9.81, the drone cannot rapidly descend which is needed in racing.

eatal, 2019-02-20:
Hi @SimpleUAVs,

I am not entirely sure what you are referring to. Currently, no minimum thrust value is set.

The achievable thrust range is defined by the allowed motor speeds. The minimum motor speed is 0 rad/s (i.e. motors can only spin in positive direction corresponding to upward thrust), this results in a minimum thrust value of 0 N.

Could you please clarify exactly what you are experiencing and where you observe the 9.81 value?

-Ezra

Winter-Guerra, 2019-02-21:
Hi @SimpleUAVs ,

Are you still having this issue? If so, would you mind please sending some more information on your issue?

Thanks,
-Winter

Winter-Guerra, 2019-02-21:
Closing due to lack of further info. Please reopen this issue if you think that this is a mistake.

closed 2019-02-21 by Winter-Guerra

SimpleUAVs, 2019-02-21:
The default “vertical” value is 9.81, which is WAY to high. Need to be zero. We’ve replaced teleop with our own code that goes to zero, and support an rc controller.

 

jz

SimpleUAVs, 2019-02-22:
Any chance of changing the throttle to a +-force, ie. Proportional throttle, from an acceleration with 9.81 as zero.

 

jz