mit-aera/FlightGoggles #62: flightgogglesRenderer process has died

parang17, 2019-02-20:
I have an error regarding rendering issue. Rviz is opened, but it said 'no image'. Here is my error log.

process[uav/world_ned_link-11]: started with pid [28787]
XIO:  fatal IO error 0 (Success) on X server ":0"
      after 223 requests (221 known processed) with 30 events remaining.
XIO:  fatal IO error 0 (Success) on X server ":0"
      after 223 requests (221 known processed) with 30 events remaining.
[uav/flightgogglesRenderer-6] process has died [pid 28670, exit code 1, cmd /home/youngjun/catkin_ws/devel/lib/flightgoggles/FlightGoggles.x86_64 -screen-quality Fastest __name:=flightgogglesRenderer __log:=/home/youngjun/.ros/log/9670464c-3555-11e9-8a93-b46bfc8eebee/uav-flightgogglesRenderer-6.log].
log file: /home/youngjun/.ros/log/9670464c-3555-11e9-8a93-b46bfc8eebee/uav-flightgogglesRenderer-6*.log
[ WARN] [1550697797.209568498]: Shutdown request received.
[ WARN] [1550697797.209612250]: Reason given for shutdown: [new node registered with same name]
[ WARN] [1550697797.211955161]: Shutdown request received.
[ WARN] [1550697797.212002799]: Reason given for shutdown: [new node registered with same name]
[ WARN] [1550697797.212519120]: Shutdown request received.
[ WARN] [1550697797.212564049]: Reason given for shutdown: [new node registered with same name]
[control_nodes/keyboard-3] process has finished cleanly
log file: /home/youngjun/.ros/log/9670464c-3555-11e9-8a93-b46bfc8eebee/control_nodes-keyboard-3*.log
terminate called without an active exception
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.
[control_nodes/universal_teleop-4] process has finished cleanly
log file: /home/youngjun/.ros/log/9670464c-3555-11e9-8a93-b46bfc8eebee/control_nodes-universal_teleop-4*.log
[uav/flightgoggles_uav_dynamics-5] process has finished cleanly
log file: /home/youngjun/.ros/log/9670464c-3555-11e9-8a93-b46bfc8eebee/uav-flightgoggles_uav_dynamics-5*.log
[uav/flightgoggles_ros_bridge-7] process has died [pid 28694, exit code -6, cmd /home/youngjun/catkin_ws/devel/lib/flightgoggles_ros_bridge/ROSClient __name:=flightgoggles_ros_bridge __log:=/home/youngjun/.ros/log/9670464c-3555-11e9-8a93-b46bfc8eebee/uav-flightgoggles_ros_bridge-7.log].
log file: /home/youngjun/.ros/log/9670464c-3555-11e9-8a93-b46bfc8eebee/uav-flightgoggles_ros_bridge-7*.log
[uav/flightgoggles_marker_visualizer-8] process has died [pid 28715, exit code -6, cmd /home/youngjun/catkin_ws/devel/lib/flightgoggles_marker_visualizer/flightgoggles_marker_visualizer __name:=flightgoggles_marker_visualizer __log:=/home/youngjun/.ros/log/9670464c-3555-11e9-8a93-b46bfc8eebee/uav-flightgoggles_marker_visualizer-8.log].
log file: /home/youngjun/.ros/log/9670464c-3555-11e9-8a93-b46bfc8eebee/uav-flightgoggles_marker_visualizer-8*.log

varunmurali1, 2019-02-20:
We need more information to understand the problem you are facing. 

Are you running this on AWS or on a local machine? 
Did this work before? 
Which launch file are you using? 
Can you post your Player.log?

It looks like you have past processes that have not died. Can you run `rosnode kill -a` and try using the launch file again?

parang17, 2019-02-21:
1. It is on a local machine. 
2. Nope. This is first time.
3. roslaunch flightgoggles teleopExample.launch
4. I putted the log. 
5. Yes, I did. But I still have the same issue.

Thank you so much.

-------------- Here is Play.log

Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
Vulkan detection: 2
Initialize engine version: 2018.3.0f2 (6e9a27477296)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: count=23
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_incremental_present, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_relaxed_block_layout, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_sampler_ycbcr_conversion, enabled=0
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_storage_buffer_storage_class, enabled=0
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_variable_pointers, enabled=0
[Vulkan init] extensions: name=VK_EXT_external_memory_dma_buf, enabled=0
Vulkan API version 1.0.57
Vulkan vendor=[Intel] id=[8086]
Vulkan renderer=[Intel(R) HD Graphics (Coffeelake 3x8 GT2)] id=[3e9b]
Vulkan device type 1
Vulkan driverversion=[0x04800005] uint=[75497477]
Vulkan PSO: cache data not found [path=/tmp/MIT-FAST-Lab/FlightGoggles/vulkan_pso_cache.bin]
Begin MonoManager ReloadAssembly
- Completed reload, in  0.060 seconds
WARNING: Shader Unsupported: 'HDRenderPipeline/Lit' - Pass 'META' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/HDRenderPipeline/TerrainLit_Basemap' - Pass 'META' has no vertex shader
WARNING: Shader Unsupported: 'HDRenderPipeline/TerrainLit' - Pass 'META' has no vertex shader
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 1920 x 1080 @ 60 Hz
InitializeOrResetSwapChain 1024x768 samples=1
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
UnloadTime: 0.344000 ms

Winter-Guerra, 2019-02-21:
Hi @parang17,

It looks like the Vulkan render backend is trying to use the integrated Intel graphics card in your PC. Unfortunately, the Intel GPU does not meet the minimum requirements for running FlightGoggles. As stated in the FAQ, FlightGoggles requires a GPU with >= 2.4GB of VRAM. Is there another GPU present inside of your computer that you could direct Vulkan to use? Alternatively, you could run FlightGoggles as a remote renderer using AWS.

Best,
-Winter

parang17, 2019-02-21:
My laptop has GTX 1070. Can you tell me how I can use this GPU?
Thanks for your helps. I am currently preparing the Alpha Pilot Challenge, so I need to figure this out.
- Youngjun

varunmurali1, 2019-02-21:
You could set the environment variable `VK_ICD_FILENAMES` to `/usr/share/vulkan/icd.d/nvidia_icd.json` as described [here](https://wiki.archlinux.org/index.php/Vulkan).

parang17, 2019-02-21:
It works. Thanks!

closed 2019-02-21 by parang17