mit-aera/FlightGoggles #64: simulator does not start normally with latest update

juanmed, 2019-02-21:
Hello,

Short description: I just updated flightgoogles. Now, core.launch and teleopExample.launch files sometimes works and sometimes they dont. The reporter.launch  works 1 out of 12 tries (tried ~25 times). So the simulator cannot start and crashes more often than not.

While I can  work by trying several times until the simulator launches correctly, I was wondering if there are any comments to improve this behaviour.

Also, the simulation frame for reporter.launch is significantly slower than that for core.launch or teleopExample.launch files (when I am able to get them working).

On another aspect, it seems FGv.2.0.0_nomultithread is either not necessary or not working with the latest update.

I am using  Ubuntu 16.04, Intel® Core™ i5-8400 CPU @ 2.80GHz, GeForce GTX 1050 Ti/PCIe/SSE2, 8GB RAM.

Thanks,

Long description:
My latest working version was with FlightGoogles Simulation Environment v2.0.0_nomultithread as was suggested in #28. 

I just updated to the latest version.  If I start FG_v2.0.0_nomultithread first ROS crashes; If i start ROS first FGv2.0.0_nomultithread crashes, with the following messages:

ROS:
terminate called after throwing an instance of 'std::out_of_range'
  what():  key 'lidarReturn' not found
[uav/flightgoggles_ros_bridge-6] process has died [pid 23333, exit code -6, cmd /home/fer/fg_ws/devel/lib/flightgoggles_ros_bridge/ROSClient __name:=flightgoggles_ros_bridge __log:=/home/fer/.ros/log/b97befa2-358b-11e9-b690-e0d55eb54399/uav-flightgoggles_ros_bridge-6.log].
log file: /home/fer/.ros/log/b97befa2-358b-11e9-b690-e0d55eb54399/uav-flightgoggles_ros_bridge-6*.log

FG_v2.0.0_nomultithread
=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Obviously I am using use_external_renderer:=1. Following are both logs for Unity3D and ROS.

[Player.log](https://github.com/mit-fast/FlightGoggles/files/2887620/Player.log)

[roslaunch-ferlabpc-23167.log](https://github.com/mit-fast/FlightGoggles/files/2887629/roslaunch-ferlabpc-23167.log)

Now, after updating flightgoogles, I can run core.launch, reporter.launch or teleopExample.launch. But they sometimes work and sometimes dont. They fail more often than not. Here is an exemple log file 
[roslaunch-ferlabpc-3766.log](https://github.com/mit-fast/FlightGoggles/files/2887661/roslaunch-ferlabpc-3766.log).

When they fail, this is the error output in console: 

=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

[uav/flightgogglesRenderer-6] process has died [pid 3922, exit code -6, cmd /home/fer/fg_ws/devel/lib/flightgoggles/FlightGoggles.x86_64 -screen-quality Fastest __name:=flightgogglesRenderer __log:=/home/fer/.ros/log/5e17ddf6-3590-11e9-b690-e0d55eb54399/uav-flightgogglesRenderer-6.log].
log file: /home/fer/.ros/log/5e17ddf6-3590-11e9-b690-e0d55eb54399/uav-flightgogglesRenderer-6*.log

Further comments will be appreciated.

Winter-Guerra, 2019-02-21:
Hi @juanmed ,

With the latest commit to master, you should use the v2.0.3 FlightGoggles binary packaged via catkin build. A mismatch between ROS code version and binary version are causing the key error you’re describing. All binaries > v2.0.0 are built without the experimental renderering pipeline and therefore should not suffer from the issues in #28. [Here are instructions](
https://github.com/mit-fast/FlightGoggles/wiki/Updating-to-Latest-Version) for syncing up the code and binary versions.

Could you please attach a player.log for a crash with the v.2.0.3 binary? The one attached does not seem to be from v2.0.3.

Best,
-Winter

Winter-Guerra, 2019-02-21:
Additionally, please post the stderr logs from your launches. The current logs seem to contain only stdout (no error dumps).

varunmurali1, 2019-02-21:
Are you always running a reporter? The reporter.launch will attempt to kill all nodes, when the reporter exits in any of the states (Completed, crashed, etc...). The set of logs that you just added tell me that the reporter timed out (i.e. the time required to complete the challenge had passed). Perhaps this is what you are seeing as a crash? Also, it is possible that some nodes are running in the background. Please try running `rosnode kill -a` before running the launch file if you suspect this is possible.

juanmed, 2019-02-21:
Hello,

I am not experiencing this issue anymore but I have no idea why. I will close this for the moment.

closed 2019-02-21 by juanmed

varunmurali1, 2019-02-21:
Hi @juanmed, This is a bit of a wild guess but my suspicion is that you have rosnodes running in the background when you experience this. If you see this again can you run `rosnode kill -a` and try the launch again?

juanmed, 2019-02-21:
Hello @varunmurali1. Thanks for the suggestion. If I experience this again I will try and let you know over here.