mit-aera/FlightGoggles #65: Need AWS AMI Image with Flight Goggles Installed

ashokyannam, 2019-02-21:
We are unsuccessful in bringing up Flight goggles in the recommended AWS EC2 instance. Same with many teams. So, I believe the best approach would be to provide AMI with all the required installations. This will help all the teams and we can focus on actual algorithms and challenge.

Winter-Guerra, 2019-02-21:
Hi @ashokyannam ,

Thank you for your feature request! We’ll take your request into consideration and we will update you with any changes.

I’m terribly sorry that your team seems to be having issues with the AWS install of FlightGoggles. Would you mind please posting a new issue with your specific problem?

Thanks,
-Winter

ashokyannam, 2019-02-21:
Awesome Winter. Majority of the teams will be very thankful to you. This would be a big relief. We will post specific issues.

BhavyanshM, 2019-02-21:
> Awesome Winter. Majority of the teams will be very thankful to you. This would be a big relief. We will post specific issues.

Oh gosh yes. That would be so amazing for so many teams out there. I was trying to achieve the same with Docker but unfortunately the trail ended in Vulkan API not being supported on Docker. Didn't think about AWS. @Winter-Guerra It would be really great if we could get a fully baked AWS AMI.

vyerneni17, 2019-02-21:
This is exactly what I've been struggling with. Tried a number of approaches and numerous attempts at building this AMI with flight goggles and all the necessary dependencies. Haven't had much success, providing this would really ease up the development process.

ashokyannam, 2019-02-25:
@Winter-Guerra shall we know ETA on this?

varunmurali1, 2019-02-25:
For now, we have chosen to make it a priority that FlightGoggles can be run locally by as many people as possible. In order to do so we are working on some optimizations that will lead to lower minimum system requirements that will enable FlightGoggles to be run on most GPUs with at least 1.7 GB of VRAM. This means that many of the common Vulkan supported GPUs with 2GB VRAM will also be supported (see #75).

Providing and supporting a well-tested AWS AMI is definitely an enhancement of value. However, at this point we cannot commit to a timeline. We'll keep the issue open and keep you up to date of any developments.

If you are having specific problems with installing/ using FlightGoggles, please open a new issue with the specific problem and we can help address it.

edowson, 2019-04-14:
See #131 for a solution to running flightgoggles from within a NVIDIA GPU accelerated docker container.