mit-aera/FlightGoggles #67: IMPORTANT: Bug in Quadrotor Dynamics Quaternion Integration

foehnph, 2019-02-21:
Hi all

I want to point out a major bug in the quadrotor dynamics, namely in file `flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp`, line 297-301:

You do quaternion integration by calculating attitudeDer (derivative of quaternion) and multiplying it with the timestep.
However you miss a factor of `0.5` from the discrete quaternion integration given by:
q_dot = 0.5 * Skew(omega) * q

The code (line 279-301)
```
double attitudeDer[4];
attitudeDer[0] = angVelocity_[0]*attitude_[3] + angVelocity_[2]*attitude_[1] - angVelocity_[1]*attitude_[2];
attitudeDer[1] = angVelocity_[1]*attitude_[3] - angVelocity_[2]*attitude_[0] + angVelocity_[0]*attitude_[2];
attitudeDer[2] = angVelocity_[2]*attitude_[3] + angVelocity_[1]*attitude_[0] - angVelocity_[0]*attitude_[1];
attitudeDer[3] = -angVelocity_[0]*attitude_[0] - angVelocity_[1]*attitude_[1] - angVelocity_[2]*attitude_[2];
```

should be
```
double attitudeDer[4];
attitudeDer[0] = 0.5* (angVelocity_[0]*attitude_[3] + angVelocity_[2]*attitude_[1] - angVelocity_[1]*attitude_[2]);
attitudeDer[1] = 0.5* (angVelocity_[1]*attitude_[3] - angVelocity_[2]*attitude_[0] + angVelocity_[0]*attitude_[2]);
attitudeDer[2] = 0.5* (angVelocity_[2]*attitude_[3] + angVelocity_[1]*attitude_[0] - angVelocity_[0]*attitude_[1]);
attitudeDer[3] = 0.5* (-angVelocity_[0]*attitude_[0] - angVelocity_[1]*attitude_[1] - angVelocity_[2]*attitude_[2]);
```

I tried to push to a new branch and do a pull request but had insufficient access rights.
Please fix this so that other teams don't struggle with wrong/funny dynamics, as we did.

eatal, 2019-02-21:
Hi @foehnph,

You are indeed correct: the factor of 0.5 is missing. We will release a patch to the issue shortly.
Thanks for bringing this to our attention.

eatal, 2019-02-21:
@foehnph Please see the recent commit 6dead2443d7bfdc245e5bc67c7210ba92917f4e6.
Thanks again
-Ezra

closed 2019-02-21 by eatal