mit-aera/FlightGoggles #68: teleopExample.launch fails

BrianN92, 2019-02-21:
I tried to use keyboard to control the flight, but it was so easy to loss its connection... Did anyone run into this issue? Thanks!

[image]

varunmurali1, 2019-02-21:
Please see this #66, we recommend that you are up to date with the master branch. The splash screen reads v2.0.1 which indicates to me that you are not up to date with the master branch.

BrianN92, 2019-02-21:
Thanks @varunmurali1 !

closed 2019-02-21 by BrianN92

varunmurali1, 2019-02-21:
PS: the keyboard teleop window needs to be focus, if you click on something else it loses focus and won't be listening to your keyboard anymore. Please see the [wiki] (https://github.com/mit-fast/FlightGoggles/wiki/Default-Controls) for more information.