mit-aera/FlightGoggles #69: Compilation error in flightgoggles_ros_bridge

mgandhi635, 2019-02-21:
Hello,

I am attempting to build this library for the first time, and I ran into a compiler error in ROSClient.cpp:185:72. The error is "invalid initialization of reference of type ‘const TransformStamped&"

Attached is the relevant log file that was produced during compilation.

Let me know if I am missing something obvious, and I appreciate the help!

[build.make.000.log](https://github.com/mit-fast/FlightGoggles/files/2890116/build.make.000.log)

varunmurali1, 2019-02-21:
Could you please add this information.

What version of ROS are you using? 
What version of Ubuntu are you using?

Winter-Guerra, 2019-02-21:
Hi there,

Could you try deleting `.tranform` from  https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles_ros_bridge/src/ROSClient/ROSClient.cpp#L185  and try compiling again?

If it still fails, please post your GCC and CMake versions.

mgandhi635, 2019-02-21:
That did it! So simple.

Thanks for the quick response.

Winter-Guerra, 2019-02-21:
Great! Glad it helped!

Winter-Guerra, 2019-02-21:
Update: patched in master branch.

closed 2019-02-21 by Winter-Guerra