mit-aera/FlightGoggles #70: PID gains after model modifications

adrianomcr, 2019-02-21:
Hello,

I've noticed that since the last updates on the quaternion integration (0.5*attitudeDer) the dynamics of the closed loop UAV has changed. As it was expected.
Wouldn't be interesting to recalibrate the gains of the PID low level controllers? So that the alphapilot teams can work with a better controlled system.

Thanks

Winter-Guerra, 2019-02-21:
Hi @adrianomcr,

The PID controller tracks angular rate commands. The vehicle attitude has no influence on it, so the fix to #67 does not affect the PID gains.

If the drone feels "sluggish" when using teleop, you can change the teleop command magnitudes by adding the following code (with the gains of your preference) to the [drone.yaml](https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles/config/drone/drone.yaml) parameter server file
```
/control_nodes:
   scales: { roll: 6., pitch: 6., yaw: 6., vertical: 19.62}
```

Regarding the PID gains: it is true that they influence the dynamic response of the drone. For that reason all of them are exposed in the [parameter server](https://github.com/mit-fast/FlightGoggles/wiki/ros-params) so that they can be changed easily.

Winter-Guerra, 2019-02-23:
Closing, since this issue is no longer relevant.

closed 2019-02-23 by Winter-Guerra