mit-aera/FlightGoggles #71: Graphics library does not with recommended Nvidia drivers

Usama1002, 2019-02-22:
Hello, I am posting this issue again. Last time, Winter G

In reference to issue#54, I was not able to resolve the problem. The FlightGoggles ReadMe says that NVIDIA driver version >=384.130 is required. But the graphics library does not work with these drivers. Uninstalling the drivers from the system lets FG start, but there is no image rendering.

I would like to know which version of OpenGL library makes the FG work correctly.

Issue#54 is pasted below for reference.

> I am trying to run FG on Ubuntu 16 having Titan X GPU with NVIDIA drivers 384.13. I was able to build it successfully. But running it throws the following error:
> 
> > [uav/flightgogglesRenderer-6] process has died [pid 3020, exit code 1, cmd /home/usr/catkin_ws/devel/lib/flightgoggles/FlightGoggles.x86_64 -screen-quality
> 
> It also gives the following error. I think the problem is related to graphics, but could not find any solution.
> 
> > rviz::RenderSystem: error creating render window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/GLX/OgreGLXContext.cpp (line 61)
> 
> Any help will be appreciated

varunmurali1, 2019-02-22:
Your error messages indicates that your base ROS install does not work. I have a few questions. 

Are you using this computer over ssh?
Are you using a virtual machine to run linux?
What graphics card is installed on your computer? 

Ignoring FlightGoggles for the moment are you able to run all of these commands successfully? 

`glxgears` should pop up a window that looks like [this](https://www.google.com/search?q=glxgears&client=ubuntu&hs=A6G&channel=fs&source=lnms&tbm=isch&sa=X&ved=0ahUKEwj6mYLQ5s_gAhUST98KHQLSCnYQ_AUIDygC&biw=1688&bih=1314#imgrc=AbpzLLJE9742bM:)
`roscore > /dev/null & rosrun rviz rviz` should open RVIZ

Winter-Guerra, 2019-02-23:
@Usama1002, do you have any updates? If not, I'm going to close this issue. The issue that you're seeing seems to be related to your particular ROS and driver installation, not an issue with FlightGoggles itself.

Usama1002, 2019-02-24:
Thanks, I am still figuring out. But I am closing the issue now.

closed 2019-02-24 by Usama1002