mit-aera/FlightGoggles #72: /image_rect_color is double-published to when running stereoTeleopExample.launch

paulerikf, 2019-02-22:
Not sure if this is worth an issue... But when using *stereoTeleopExample.launch*:  `/uav/flightgoggles_ros_bridge` publishes to the `.../image_rect_color` topics, while stereo_image_proc both subscribes to and publishes to them as well.  

[image]

Winter-Guerra, 2019-02-22:
Good catch! I think this is happening because `stereo_image_proc` is rectifying the already rectified image and then publishing the rectified image back. This doesn’t really harm the end user, but is good to be aware of. I’ll see if I can fix this in next patch (probably by using a `/dev/null` disposal topic).

Winter-Guerra, 2019-03-08:
Resolved in master. Thanks for the bug report!

closed 2019-03-08 by Winter-Guerra