mit-aera/FlightGoggles #75: Renderer fails on GTX760. Kind request for 2GB VRAM version.

pieniacy, 2019-02-24:
As already stated, 2.1GB VRAM is required to run renderer for now. It seems like being just a tiny little bit over many popular graphics cards with 2GB of VRAM. This issue is mostly a feature request for a small optimization or compromise on quality so it fits on so popular 2GB cards. Many thanks in advance.

Winter-Guerra, 2019-02-25:
Hi @pieniacy,

Thanks for the feature request! Based on some of my experimentation from Friday, it _might_ be possible to lower the bare-minimum VRAM requirement to ~1.7GB using texture streaming. The environment itself is > 10GB in its raw form, which makes lowering VRAM usage quite hard. I’ll test this a bit more and hopefully integrate it into a new patch release. However, I can’t make any promises about it. 

Best,
-Winter

Winter-Guerra, 2019-02-27:
Update: I was able to make a build today with 1.5GB and 1.7GB VRAM requirements. As expected, this feature was very hard to implement. It took a lot of render pipeline trickery to make everything fit into the VRAM allotted.

I'm going to clean up the build and will release it sometime tomorrow (Wednesday).

Winter-Guerra, 2019-02-28:
Hi @pieniacy,

The new (beta) release of the binary with reduced VRAM requirements can be found [here](https://github.com/mit-fast/FlightGoggles/releases/tag/v2.0.4-b1). Please read the release notes for usage instructions and advisories on known issues.

Sorry for the wait!

Best,
-Winter

pieniacy, 2019-02-28:
Thanks! I personally am already developing on 4gigs card, but it surely will enlarge number of team members able to run the sim. We should be able to test it today and will be back with results asap.

pieniacy, 2019-02-28:
We successfully tested sim on a notebook with 2GB VRAM! Looks like You did great job optimizing it :)

Here are specs for reference:
```
RAM 15,6 GiB
Intel® Core™ i7-3610QM CPU @ 2.30GHz × 8 
GeForce GTX 660M/PCIe/SSE2
```
Thank you!

closed 2019-02-28 by pieniacy

aj2020tls, 2019-04-06:
Hi Winter,

Can you give us a GPU footprint for the very high and ultra setting ?

Regards,
-andy

nilay994, 2019-06-06:
Hi Winter, 

Would there be any way to still go lower on the requirements of the renderer? The current documentation does show a way [to reduce the resolution](https://github.com/mit-fast/FlightGoggles/wiki/Unity3D-Usage_Guide) of the simulator - but seems like it requires an installation of Unity. Does this reduce the VRAM requirements of the sim? If yes, can the resolution be lowered without installing Unity?