mit-aera/FlightGoggles #78: Reporter Node Dies

PHRABAL, 2019-02-25:
Hi,

I'm running FG successfully for a couple weeks. Use `core.launch` to get it going no problems. However, I wanted to run one of the practice challenges today to test my trained drone and when I do `roslaunch flightgoggles reporter.launch level:=easy` I see the core launches successfully, states the initial pose, but then:

```
Traceback (most recent call last):
  File "/home/ubuntu/catkin_ws/src/flightgoggles/flightgoggles_reporter/src/reporter.py", line 9, in <module>
    import tf
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
[reporter-12] process has died [pid 2759, exit code 1, cmd /home/ubuntu/catkin_ws/src/flightgoggles/flightgoggles_reporter/src/reporter.py __name:=reporter __log:=/home/ubuntu/.ros/log/a8486b42-3956-11e9-986e-0229bea2e0ac/reporter-12.log].
log file: /home/ubuntu/.ros/log/a8486b42-3956-11e9-986e-0229bea2e0ac/reporter-12*.log
```

Winter-Guerra, 2019-02-25:
Could you please try changing [this line](https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles_reporter/src/reporter.py#L1) to `#!/usr/bin/env python2`? ROS TF does not work with python 3 and will throw a similar error.

PHRABAL, 2019-02-26:
This eliminated the error. However, I launch the reporter and then try in my python code to publish commands as well as retrieve images from the sim and it doesn't work. For example, when trying to publish a command, I can't get a connection. When I try to grab an image, there are none, although when I `rostopic echo` data is there. Does the reporter node somehow change what my working Python code has to do to send and receive?

Winter-Guerra, 2019-02-26:
Could you please post the specific error that you're getting? `reporter.py` only subscribes to topics. It should not effect any of your code or ROS topics. 

It's possible that you're in a python3 virtual environment that is not setup for ROS. Could you also please also post your RQT graph? It's possible that your nodes/topics are not wired up correctly.

PHRABAL, 2019-02-26:
I want to dig into the `reporter.py` code and retry some things and see if I can find the source myself. If I get stuck, I'll open a new issue with the info you requested. Since you solved the original error, you could close this issue if you want. Thank you guys for replying so fast to our issues. Deadline is tight for this competition. It's very much appreciated!

Winter-Guerra, 2019-02-26:
No problem, best of luck!

closed 2019-02-26 by Winter-Guerra