mit-aera/FlightGoggles #79: New Unity Renderer freezing my computer

DevMohamedMI, 2019-02-26:
We're now using the newly released renderer v2.0.3, and the latest commit of the Flightgoggles repository. Freezing occurs when we launch core.launch with flag render_stereo:=true. Previously we were using the v2.0.0-no_multi_thread such as linked in this #28  thread. The multi-thread original renderer was crashing on the hardware listed below as well. Could you release a no-multi-threaded v2.0.3, or some other fix? 

My system is:
AMD RYZEN Threadripper 1900X 8-Core / 16
GeForce RTX 2080
NVIDIA-SMI 410.79
Ubuntu 16.04.5 LTS

Winter-Guerra, 2019-02-26:
Hi @DevMohamedMI,

The v2.0.3 renderer does not use the experimental pipeline. Your crash is probably related to rendering stereo. The FAQ has a section on rendering stereo and probably explains why you are having this issue. This issue is related to #51 (closed), which you can also take a look at for more info.

> Q. Does FlightGoggles currently support Stereo?
> Yes! Simply pass `render_stereo:=true` to the FlightGoggles launch file. Note: as of `v2.0.2` the combined vertical resolution of the two stereo cameras must not be greater than the vertical resolution of the largest monitor attached to your computer. In other words, you must have a monitor with `>=1536` vertical resolution to render stereo at the default camera resolution. A current workaround is to [decrease the default camera resolution](https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles/config/drone/drone.yaml#L47), attach a high resolution "dummy" monitor, or rotate or attach a vertical display to your rendering computer.

> As an explanation for the current restriction, FlightGoggles uses the GPU backbuffer for storing rendered images. However, the size of this buffer is limited by the screen resolution. Thus, concatenated images cannot be larger than the size of the GPU backbuffer. This issue will be patched in a later release.

Best,
-Winter

DevMohamedMI, 2019-02-26:
Alright thanks so much @Winter-Guerra !

closed 2019-02-26 by DevMohamedMI