mit-aera/FlightGoggles #80: reporter.py crashing

SimpleUAVs, 2019-02-26:
reporter.py crashes on line 121 with error "tuple index out of range" and a race condition error during launch of challenge.

LOG:

[reporter-12.txt](https://github.com/mit-fast/FlightGoggles/files/2903265/reporter-12.txt)

Winter-Guerra, 2019-02-26:
How are you running the script? The level arg in the reporter.launch needs to be specified. It is clear from your error that the level params are not being loaded. Try `roslaunch flightgoggles reporter.launch level:=hard`

SimpleUAVs, 2019-02-26:
using the challenge launch shell file rosrun flightgoggles scorer.sh. use "roslaunch flightgoggles scorer.launch for repro.

Winter-Guerra, 2019-02-26:
OK, we were able to reproduce the problem using the AlphaPilot challenge files. Here's how we fixed it.

- The wget link in `README_v2.md` is broken. Instead, use `wget https://d253pvgap36xx8.cloudfront.net/challenges/resources/0e5a78da357911e9b4000242ac110002/Challenge_Leaderboardtest.zip`
- In `scorer.launch`, there is a missing namespace around the rosparam loads. The block starting at line 37 should look like the following:

```xml
<group ns="/uav">
  <rosparam command="load" file="$(find flightgoggles)/config/challenges/gate_locations_$(arg gate_locations).yaml"/>
  <rosparam command="load" file="$(find flightgoggles)/config/challenges/challenge_final.yaml"/>  
</group>
```

- In `scorer.sh`, `use_external_renderer:=1` on line 8 should be removed.

Best,
-Winter

Winter-Guerra, 2019-02-26:
Closing since the solution here seems to fix the issue.

closed 2019-02-26 by Winter-Guerra