mit-aera/FlightGoggles #81: FlightGoggles renderar (Unity) crashes on Ubunto 16.04 LTS, Dell T3600, Nvidia 1080i w/ driver vs418

SimpleUAVs, 2019-02-26:
Unity is crashing on startup on above configuration with "Shader Unsupported: 'HDRenderPipline/LitTessellation'" error. Player.log attached

[Player.txt](https://github.com/mit-fast/FlightGoggles/files/2904185/Player.txt)

Winter-Guerra, 2019-02-26:
Those shader warnings are normal and probably not the cause of the crash. Linux is not able to render all of the effects in the environment.

We will need a bit more info to diagnose this issue. Does `vulkan-smoketest` run? Are you running this inside of docker? Does this crash every time? Are you running with stereo?

SimpleUAVs, 2019-02-26:
Yes to vulkan-smoketest, no to docker, no to everytime, no to stereo. It's a new install from scratch on raw iron.

edowson, 2019-04-13:
Have you edited `/usr/share/vulkan/icd.d/nvidia_icd.json` and renamed the library path entry from `libGL.so.1` to `libGLX_nvidia.so.0` ?

#131

closed 2019-07-17 by Winter-Guerra