mit-aera/FlightGoggles #82: IR Markers inconsistent?

foehnph, 2019-02-26:
Hi all

I have a question regarding the IR markers.
It seems that the marker positions from the "nominal_gate_locations.yaml" do not coincide with the corner locations used for rendering the IR markers.
I do not mean the difference of the nominal gate position and the perturbed (in the context of AlphaPilot), but effectively the corners seem to be wrong for some gate.
If we calculate the reprojections for certain gates (i.e. Gate21), they seem to be wider/taller/smaller than given by the corner information from the yaml file.
Our reprojection formula/algorithm appears to be correct since it agrees with most of the gates (from a variety of viewpoints) and their corner IR markers from FlightGoggles, but not with Gate21 and some others....

Winter-Guerra, 2019-02-26:
Hi @foehnph,

We're looking into this now. This is an odd problem since the nominal IR locations were output by FlightGoggles and should correspond to the nominal locations in the environment.

Is gate 21 the only gate that exhibits this issue? If not, which others also exhibit the issue? We're going to try to reproduce via reprojection and will get back to you. For the purposes of reproducing the issue, any examples you could provide would be very helpful.

Best,
-Winter

Winter-Guerra, 2019-02-26:
Ah, this might be a marker ID association problem stemming from a performance improvement fix introduced in [this commit.](https://github.com/mit-fast/FlightGogglesRenderer/commit/a4c2cce7336508cdfb93e9d0a56bb9c2ddeb0ae5#diff-7575e094a5a2d07db9ae1aeef24bb01d)

Will verify and put out a patch if needed.

foehnph, 2019-02-26:
I'm not concerned about the marker IDs. Our reprojections are still close to the projections from unity.
Just for some gates they seem to be distributed around the gate position with a different height/width compared to the one in the `nominal_gate_locations.yaml` (we do account for gate rotation with respect to the camera pose). 
atm I can not check for all the gates... no time right now, sorry.
I will provide an image illustrating what I mean later on

Winter-Guerra, 2019-02-26:
Ok, thanks! I'm looking forward to seeing the image(s). It would be a big help.

foehnph, 2019-02-26:
red are the gate markers given by unity, green our computed reprojection.
In this scenario there is no perturbation. Note that the markers of the closer gate are perfectly overlaying and the markers further away (gate21) are not concentric.
It happens for other gates, too and is not related to the distance as far as we can tell.

[image]

varunmurali1, 2019-02-26:
Hi @foehnph ,

It looks like some of the marker locations in `nominal_gate_locations.yaml` were from a previous beta build of FlightGoggles. We've updated the file and no longer have reprojection issues (see attached screenshot). Sorry about that...

[image]

foehnph, 2019-02-27:
perfect! this fixed the problem!

Thanks a lot!

closed 2019-02-27 by foehnph