mit-aera/FlightGoggles #83: Local client unable to communicate with AWS master

tristandeborde, 2019-02-26:
Hello, 
I am currently having trouble running FG on an AWS p3.2xlarge server with ubuntu 16.04 deep learning AMI, and streaming it to my local ubuntu 16.04 machine.

### More info on my setup:
- I have reinstalled Flightgoggles on a new AWS instance and on my local machine yesterday, from scratch (v2.0.3)
- I have already followed the solutions detailed here #37 ; I launched a VPN between my AWS instance and my machine. Thus I have two new network interfaces: 10.8.0.1 corresponds to the AWS server's IP as generated by the VPN, and 10.8.0.6 corresponds to the local machine's IP as generated by the VPN. I can confirm that the VPN is working, because I can ping the two new IPs from both the AWS instance and my local machines; and the launch commands for openvpn are not outputting any errors.
Following that, here are my ROS env variables:
- on AWS: `echo "$ROS_IP $ROS_MASTER_URI"` returns `10.8.0.1 http://10.8.0.1:11311`. 
FYI, 
- on local machine: `echo "$ROS_IP $ROS_MASTER_URI"` returns `10.8.0.6 http://10.8.0.1:11311`

### Detailed output of my commands: 
- `roslaunch flightgoggles core.launch` returns no error on AWS, and seems to be functional. Here is the output: 
```
... logging to /home/ubuntu/.ros/log/04f7de5a-39ce-11e9-b02a-0653345eb61a/roslaunch-ip-172-31-31-3-15435.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.8.0.1:45457/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /uav/Gate1/nominal_location: [[-0.009051204, -...
 * /uav/Gate1/perturbation_bound: [2.5, 2.5, 5.0]
 * /uav/Gate10/nominal_location: [[18.15111, 3.631...
 * /uav/Gate10/perturbation_bound: [2.0, 2.0, 5.0]
 * /uav/Gate11/nominal_location: [[-27.01367, 2.54...
 * /uav/Gate11/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate12/nominal_location: [[18.40823, 13.60...
 * /uav/Gate12/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate13/nominal_location: [[1.237332, 9.001...
 * /uav/Gate13/perturbation_bound: [2.5, 2.5, 5.0]
 * /uav/Gate14/nominal_location: [[-8.674376, -29....
 * /uav/Gate14/perturbation_bound: [1.0, 1.0, 5.0]
 * /uav/Gate15/nominal_location: [[5.744822, -11.7...
 * /uav/Gate15/perturbation_bound: [0.0, 2.5, 5.0]
 * /uav/Gate16/nominal_location: [[-13.73218, 28.6...
 * /uav/Gate16/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate17/nominal_location: [[-20.57, -13.298...
 * /uav/Gate17/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate18/nominal_location: [[13.51782, 21.76...
 * /uav/Gate18/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate19/nominal_location: [[15.96136, -17.2...
 * /uav/Gate19/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate2/nominal_location: [[0.8248205, 27.8...
 * /uav/Gate2/perturbation_bound: [0.0, 1.0, 5.0]
 * /uav/Gate20/nominal_location: [[-25.39118, 25.0...
 * /uav/Gate20/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate21/nominal_location: [[15.68563, 38.50...
 * /uav/Gate21/perturbation_bound: [0.0, 0.0, 2.0]
 * /uav/Gate22/nominal_location: [[4.803625, -27.3...
 * /uav/Gate22/perturbation_bound: [1.0, 1.0, 5.0]
 * /uav/Gate23/nominal_location: [[-9.328671, 7.77...
 * /uav/Gate23/perturbation_bound: [5.0, 5.0, 5.0]
 * /uav/Gate3/nominal_location: [[-0.3703381, -12...
 * /uav/Gate3/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate4/nominal_location: [[-31.56487, -13....
 * /uav/Gate4/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate5/nominal_location: [[-20.94317, 14.6...
 * /uav/Gate5/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate6/nominal_location: [[-9.14867, 30.62...
 * /uav/Gate6/perturbation_bound: [0.0, 0.0, 2.0]
 * /uav/Gate7/nominal_location: [[16.2506, 41.63,...
 * /uav/Gate7/perturbation_bound: [0.0, 0.0, 0.0]
 * /uav/Gate8/nominal_location: [[-3.430186, 11.7...
[... 101 lines cut ...]

WARNING The following node subscriptions are unconnected:
 * /uav/flightgoggles_uav_dynamics:
   * /uav/input/rateThrust
```

- BUT flightgoggles seems to be unable to communicate with the AWS master server. When `roslaunch flightgoggles core.launch` runs on AWS, launching `roswtf` on my local machine fails. This looks a lot like a connection timeout: if I set my ROS_MASTER_URI to a deliberately wrong IP, roswtf instantly outputs an error, whereas this is taking a long time to fail. Here is the output:
```
Loaded plugin tf.tfwtf
No package or stack in context
================================================================================
Static checks summary:

No errors or warnings
================================================================================

ROS Master does not appear to be running.
Online graph checks will not be run.
ROS_MASTER_URI is [http://10.8.0.1:11311]
```

- Launching `roslaunch flightgoggles core.launch` on my local machine works if my ROS env vars are unset: a GUI window opens up for flightgoggles, and the simulation starts. Since my machine is slow I am forced to close this window.

Do you have any idea on the origin of these problems ? 
Thanks in advance

Winter-Guerra, 2019-02-26:
Hi @tristandeborde,

The sort of issue you're getting sounds like a firewall issue not allowing for your local machine to contact the AWS ROS master. Have you tried adding UDP port 1194 to the allowed rules of the AWS security group? According to [this](https://serverfault.com/questions/873050/openvpn-server-doesnt-listen-to-client-through-netcat?rq=1) disabling ubuntu's firewall may also help `ufw disable`.

tristandeborde, 2019-02-26:
@Winter-Guerra Wow, disabling the firewall with `ufw disable` on both machine solved the connection problem! Thanks a lot for the insight. I can know view the simulation and control the drone, but the simulation is very "laggy" and has a low rendering quality, though.
I have down a speedtest on both my local machine and the AWS instance, and the download speed in each case is superior to 40mb/s.

Do you have any idea on how to speed up the simulation ? 
Thanks again for the very quick and accurate answer

Winter-Guerra, 2019-02-26:
Hi @tristandeborde ,

I’m glad that `ufw disable` solved the issue! 

Regarding render quality and speed, 30FPS at the default resolution `1024 x 768` is about 70MB/s. Thus, I would expect that your remote camera feed would feel slow and laggy. For this reason, we expect that AWS users would try to keep most of their algorithm development inside of AWS and only use the external stream for debugging. The simulator can also run in 'sim_time', which guarantees 60hz sim-time camera feeds, but slows down time.

I'm going to close this issue, since it seems like you got the simulator to work on AWS. Please open another if you have a separate issue with the AWS simulator.

Best,
-Winter

closed 2019-02-26 by Winter-Guerra

tristandeborde, 2019-02-26:
@Winter-Guerra You are absolutely right, after lowering the drone camera resolution in the `src/flightGoggles/flightgoggles/config/drone/drone.yaml`, the camera feed is smooth!
Thanks again for your help on this matter.