mit-aera/FlightGoggles #85: Multiple instances of an environment

koriavinash1, 2019-02-27:
Is there any method to create multiple instances of an environment? 
(Each agent interacting with different set of ROS topics specific to particular instance)

Winter-Guerra, 2019-03-01:
Hi @koriavinash1,

The latest [`v2.0.4-b1` beta build](https://github.com/mit-fast/FlightGoggles/releases/tag/v2.0.4-b1) for the FlightGoggles renderer includes CLI args for specifying connection port numbers for the ROS bridge connection. Theoretically, one could spawn multiple instances of the FlightGoggles renderer and ROS bridge in multiple namespaces for conducting multiple experiments at the same time.

However, the ROS bridge currently does not take in ROS params or CLI args for setting the [input and output connection ports](https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles_ros_bridge/src/Common/FlightGogglesClient.hpp#L41). You could try implementing this yourself, but this feature is currently unsupported. 

Best,
-Winter

koriavinash1, 2019-03-01:
Can we change ports in launch files directly? 
This feature would be helpful, for training A3C agents ...