mit-aera/FlightGoggles #86: AlphaPilot Leaderboard scorer/reporter errors - possible launch file namespaceing

foehnph, 2019-02-27:
Hi again

One more minor issue regarding the Leaderboard entry for AlphaPilot and the scorer/reporter scripts/nodes.

When we start the scripts we get errors such as:
```
[ERROR] [1551273718.200331, 0.000000]: Challenge name could not be read
[INFO] [1551273718.317813, 0.000000]: Running challenge 
[ERROR] [1551273718.318811, 0.000000]: Timeout not specified!
```
We could narrow it down to some namespace problem.
In the file `scorer.launch` lines 37/38 should be namespaced into `/uav` I guess...
```
<group ns="/uav">
<rosparam command="load" file="$(find flightgoggles)/config/challenges/gate_locations_$(arg gate_locations).yaml"/>
<rosparam command="load" file="$(find flightgoggles)/config/challenges/challenge_final.yaml"/>  
</group>
```

But then the reporter/scorer always load nmonial gate positions instead of perturbed ones for testin.

varunmurali1, 2019-02-27:
You are correct, these errors do likely indicate an issue with the namespace. This appears to be the same issue as #80. Could you try the solution there? 

Are you trying to run the scorer launch by itself? It looks like you need to pass in a `gate_locations` arg which defaults to 0 otherwise.

foehnph, 2019-02-28:
Thanks a lot and sorry for not realising this has already been posted.
In fact, correct namespacing and changing external_renderer to false solved the problem.

closed 2019-02-28 by foehnph