mit-aera/FlightGoggles #87: Need Gate Passage Event Notification

BhavyanshM, 2019-02-28:
Hello,

Could you please suggest ways to access the information about events such as successful passage through a gate? 

Thanks!

varunmurali1, 2019-02-28:
In terms of FG, the reporter is the only node that has access to these events and you can publish an event right after Line 141 in reporter.py. If your question is whether you can make this modification in the context of AlphaPilot, that is a question for AlphaPilot....

BhavyanshM, 2019-02-28:
> In terms of FG, the reporter is the only node that has access to these events and you can publish an event right after Line 141 in reporter.py. If your question is whether you can make this modification in the context of AlphaPilot, that is a question for AlphaPilot....

Oh okay, yes I just wanted to make sure. Then it's definitely a question for AlphaPilot. Thanks though.

closed 2019-02-28 by BhavyanshM