mit-aera/FlightGoggles #89: Minor Bugs in AlphaPilot Leaderboard Environment

foehnph, 2019-02-28:
There are 3 minor problems within the AlphaPilot Leaderboard Environment:

- [ ] The script `scorer.sh` indexes gates 1 to 25, while the gate files are indexed 0 to 24. This leads to a crash of the reporter in the last round (not finding index 25). changing line 3 in `scorer.sh` to `for i in 'seq 0 24';` leads to another crash of the reporter at startup without any useful error indication.

- [ ] Changing the parameter `/uav/flightgoggles_uav_dynamics/clockscale` does invoke realtime rendering, but the framerate drops significantly and the parameter itself seems to have no effect. Neither can we speed up the simulation (aware of potential performance bottlenecks), nor can we slow it down.

- [ ] For some gate perturbations, impossible situations arise. For example, the second last target gate `Gate23` ends up right in front or behind a pillar. As seen in the attached screenshot. Can you make sure such (nearly) impossible situations do not arise in the leaderboard and test tracks?

[image]

Thanks again

Winter-Guerra, 2019-03-01:
Hi @foehnph,

Unfortunately, the `scorer.sh` file is provided by the AlphaPilot challenge. We'll investigate its incompatibility with the FlightGoggles reporter and the environment gate perturbations. We'll get back to you as we have updates.

Regarding the clockscale, this issue appears to be new. We'll attempt to recreate it and will push a patch if necessary.

Best,
-Winter

eatal, 2019-03-04:
The moment of inertia is represented by a spherically symmetric model (identical inertia around each principal axis), as per [drone.yaml](https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles/config/drone/drone.yaml). Consequently, the angular rate vector and the angular momentum vector are always aligned, and thus the gyroscopic torques are zero.

Soon we will release an additional multicopter dynamics library, that is suitable for simulating vehicles with different configurations for use outside Alphapilot.

Winter-Guerra, 2019-03-14:
Hi @foehnph,

We have not been able to reproduce your issues with the clock scale in any of our testing. We can confirm that the clockscale setting is working as intended, including for clockscales faster than real time. When `/use_sim_time` is set to true, uncommenting `/uav/flightgoggles_uav_dynamics/clockscale` allows for the user to set a fixed clockscale. However, we did notice something that might cause users some slight confusion. At faster than real time clockscales, the renderer requires a few seconds of time on startup to "catch up". Additionally, setting this parameter manually will override automatic clockscaling and cause your drone's framerate to drop in sim-time, which is expected.

We're still working with AlphaPilot to fix items 1 & 3 on your list. For the moment, all users are encouraged to use the fixes listed in #80 until AlphaPilot releases an updated version of the scripts and gate perturbations.

Best,
-Winter & Varun

closed 2019-03-15 by Winter-Guerra