mit-aera/FlightGoggles #91: ir_beacons marker_ID format

ashwinvk94, 2019-02-28:
For this issue, I am assuming that the quad is starting from final_challenge init position - [18, -23, 5.3, 0.0, 0.0, 0.707, 0.707].

**Note**: In order for discussion in this issue, I am ordering the ir_beacons from the top_left corner and then in the clockwise direction.

If we fly forward from this position, we observed that the **gate 10** corners are arranged in the order **[2,1,3,4]** But the next gate(**gate 21**) is ordered in the order **[1,2,4,3]**.

Is this a bug, or is there no order to the marker_ID's for all the gates.

Winter-Guerra, 2019-02-28:
Hi @ashwinvk94 ,

The ROS param marker ordering and `ir_beacon` IDs should match and should be consistent (see attached photo). We guarantee that the association between ROS param marker order and `ir_marker` IDs are consistent, but do not guarantee that all gates will have a clockwise ordering in the direction of the racing line. Please make sure to update your git repo to the latest version to avoid issue #82.

Best,
-Winter
 
[image]

closed 2019-02-28 by Winter-Guerra

pieniacy, 2019-03-01:
@Winter-Guerra are they consistent though? Clockwise or counterclockwise difference is understandable, but in even in your picture we can see that numbering is inconsistent: diagonals can be 1-4, 2-3 or 1-3, 2-4 depending on gate. It would be nice if we had always 3 under 1 or something like this.

Winter-Guerra, 2019-03-02:
@pieniacy,
By inspection, it looks like square gates have 3 under 1, all others (circular and post-based) have 4 under 1. We originally meant for all to have 4 under 1, but are now grandfathered into this particular arrangement (sorry!). This arrangement will not change going forward.

Best,
-Winter