mit-aera/FlightGoggles #94: Pausing simulator

Cpruce, 2019-03-01:
Hi,

Is there a way to pause/unpause the simulator through ros? I see /uav/collision is for reseting the simulator. I didn't see a topic to pause though. If one doesn't exist, do you have suggestions for adding that functionality?

Thanks,
Cory

varunmurali1, 2019-03-04:
I am marking this as a feature request. However, it is unlikely we be providing this functionality soon. If you wish to implement this yourself, I would look at the way the simulationLoopTimer works in the [dynamics node] (https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp). In theory, you could write a callback that stops this timer (on pause) and starts it on unpause. Note, this would only work in simulation time and you might have to edit the logic in line 230 of that file.

Winter-Guerra, 2019-07-17:
Closing this issue, since this is a feature that we're unlikely to implement as part of the core FlightGoggles stack.

closed 2019-07-17 by Winter-Guerra