mit-aera/FlightGoggles #96: CHANGE_REQUEST: Update UAV flight dynamics to be more representative of IRL quad drone racer

SimpleUAVs, 2019-03-04:
We did some testing with our MultiGP pro racers. All testing was done with our control system (joy convert to rateThrust) with a FrSky Terranis on a Dell T3600 E5-1660 w/Sea Hawk 1080i. Comments below:

1-Way to much inertia in all axes. A racing drone is incredibly agile, the sim is VERY sluggish. Leveling after accelerating drone maintains speed. It should slow quickly and stop. Tight turns end up being skids, drifts. etc
2. The does not respond to thrust vectoring, ie high bank angles and using the throttle to pull through corners. increasing throttle just increases altitude, not speed or decrease turn radius as it should.
3. Roll and yaw rates are way to low. Several corners on the course are almost impossible to make at speed with current settings.
4. Camera angle way to low. At speed, gate locations have to be guessed because they cannot be seen. The camera angle determines the min and max speed.
5. Turn from gate (in order, not name) 5 to 6 to 7 is extremely difficult due to rates and inertia carried into the turn. Our best pilots had only made this turn 50% of the time.
6. Gate 2 to 3 requires a lot of lead, ie turn has to start a long time before passing gate 2.
7. Reverse expo on all channels help a bit, but not enough
8. GPU was hardly taxed by the resulting in lower than expected frame rates. Temp of GPU was 45-55 deg C when it should have been 70-80 deg C. Frame rates should have been 60-120. Frame rate was 23 in real time, 60 sim time.

I suggest that MIT consult with a pro racer on flight dynamics because right now FlightGoggles isn't even close. All our pilots compared it to the early, hacked together sims of 5 years ago.

Video of test session is available but is quite long. I'll post excerpts to youtube soon. Let me know of any parts you want focus on.

varunmurali1, 2019-03-05:
All parameters of the drone are exposed and can be changed by the end user. 

Yes, some of your comments do indeed apply to FPV racing drones that are available on the market. However, there seems to be a misconception on the dynamics of an _autonomous_ drone, which our parameters are based on. In particular, autonomy requires a powerful on-board computer, such as the NVIDIA Jetson Xavier, which alone weighs more than 650 grams. 

The framerates you mention are expected behavior. The camera rate is set to 60Hz in sim time (and capped to it in real time). Please note that, after the rendering, the rendered images need to be transferred out of the GPU memory into the ROS framework, which adds additional time to the simulation. 

Finally, it seems like a number of your comments pertain to the AlphaPilot challenge. Please use the herox forum for such questions.

closed 2019-03-05 by varunmurali1

SimpleUAVs, 2019-03-06:
I am not talking about what is available on the market. No one flies a pre-built in competition, not if the want to be competitive. All the racers I know custom build their own as do we. The AlphaPilot program, and particularly DRls involvement states that it is to end in a race against an IRL pilot, thus the dynamics used to train on must be as accurate, and as true to life as possible. I know most of the DRL pilots, members of my team and club have raced against them, and beaten them. We had three at our regional final.

Autonomy for drone racing does not require a powerful onboard computer. We had some success two years ago with an Arduino on a telemetered track. Multiple stereo cameras and a lidar are definitely not required. Weight is the enemy. Be as light and agile as possible which the sim is definitely not. Most competitive racers, at least at this year's finals, are under 250g w/o battery (6 cell) and push more than 2000w.

Not saying the sim is bad, just that it's not suited for racing or training racers. But it can be. If you really want to know how, contact me.

And your team has been great with support of FlightGoggles. Wish I could say the same about the AlphaPilot team.

FWIW, the video is at https://youtu.be/cFEBYTdiPUw . ROSBags are available.

SimpleUAVs, 2019-03-06:
I am not talking about what is available on the market. No one flies a pre-built in competition, not if the want to be competitive. All the racers I know custom build their own as do we. The AlphaPilot program, and particularly DRls involvement states that it is to end in a race against an IRL pilot, thus the dynamics used to train on must be as accurate, and as true to life as possible. I know most of the DRL pilots, members of my team and club have raced against them, and beaten them. We had three at our regional final.

Autonomy for drone racing does not require a powerful onboard computer. We had some success two years ago with an Arduino on a telemetered track. Multiple stereo cameras and a lidar are definitely not required. Weight is the enemy. Be as light and agile as possible which the sim is definitely not. Most competitive racers, at least at this year's finals, are under 250g w/o battery (6 cell) and push more than 2000w.

Not saying the sim is bad, just that it's not suited for racing or training racers. But it can be. If you really want to know how, contact me.

And your team has been great with support of FlightGoggles. Wish I could say the same about the AlphaPilot team.

FWIW, the video is at  <https://youtu.be/cFEBYTdiPUw> https://youtu.be/cFEBYTdiPUw . ROSBags are available.