mit-aera/FlightGoggles #100: BUILD_BREAK: Latest FlightGoggle system does not build under catkin_make on system previous builds did

SimpleUAVs, 2019-03-10:
Build Break with following error:

CMake Warning at catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on '/usr/lib/x86_64-linux-gnu/libSDLmain.a' but
  neither '/usr/lib/x86_64-linux-gnu/libSDLmain.a_INCLUDE_DIRS' nor
  '/usr/lib/x86_64-linux-gnu/libSDLmain.a_LIBRARIES' is defined.
Call Stack (most recent call first):
  catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  keyboard/CMakeLists.txt:26 (catkin_package)

CMake Warning at catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on '/usr/lib/x86_64-linux-gnu/libSDL.so' but
  neither '/usr/lib/x86_64-linux-gnu/libSDL.so_INCLUDE_DIRS' nor
  '/usr/lib/x86_64-linux-gnu/libSDL.so_LIBRARIES' is defined.
Call Stack (most recent call first):
  catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  keyboard/CMakeLists.txt:26 (catkin_package)

CMake Warning at catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on '-lpthread' but neither
  '-lpthread_INCLUDE_DIRS' nor '-lpthread_LIBRARIES' is defined.
Call Stack (most recent call first):
  catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  keyboard/CMakeLists.txt:26 (catkin_package)

-- +++ processing catkin package: 'rosbaglive'
-- ==> add_subdirectory(rosbaglive)
-- +++ processing catkin package: 'agsap'
-- ==> add_subdirectory(agsap)
CMake Warning at /home/ags/catkin_ws/devel/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "opencv" with any
  of the following names:

    opencvConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "opencv" to CMAKE_PREFIX_PATH or set
  "opencv_DIR" to a directory containing one of the above files.  If "opencv"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  agsap/CMakeLists.txt:10 (find_package)

-- Could not find the required component 'opencv'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /home/ags/catkin_ws/devel/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "opencv" with any
  of the following names:

    opencvConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "opencv" to CMAKE_PREFIX_PATH or set
  "opencv_DIR" to a directory containing one of the above files.  If "opencv"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  agsap/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ags/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ags/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:2754: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
ags@ags02:~/catkin_ws$ 

System is Dell t3600, E5-1660 Xeon, Sea Hawk NVidia 1080i, Ubunto 160.04 LTS, OpenCV 4.0.1, Python 2.7 and 2.5.2. 

It appears the BOM and dependencies changed. If so, please document.

jz

Winter-Guerra, 2019-03-11:
Hey Jeff,

The CMake errors that you're seeing are coming from your `agsap` package. This seems like a package that you're providing and is not related to FlightGoggles. There are some warnings regarding the `keyboard` package, but they can be ignored and are not effecting your build. 

Closing issue since this error is not related to the FlightGoggles framework.

closed 2019-03-11 by Winter-Guerra

SimpleUAVs, 2019-03-11:
No they are not. We have no dependencies on libSDL, And aGSP is ALL PYTHON and runs just fine.

SimpleUAVs, 2019-03-11:
Then why did it build this morning just fine and not after I did a git pull on your code? Flightgoggles updates were the ONLY delta.

SimpleUAVs, 2019-03-11:
Still fails with agsap project removed!

Error at 97%, MakeFile:138

engcang, 2020-03-10:
@Winter-Guerra @SimpleUAVs 
How are you doing Winter? Long time no see!
This is EungChang from KAIST, who participated in AlphaPilot last year.

I suffered the similar issue with regard to "libSDL.so, libSDLmain.a" on my new two desktops and solved it.
I hope this can help the others who could experience same issue and can help you to figure it out maybe.

I simply
$ cd <workspace> && catkin clean -y
$ cd <workspace>/src/flightgoggles
$ git checkout 8a0af6c #28th of May, 2019 version of your repo.
$ cd <workspace> && catkin build

I verified this several times by checking out to recent commit version, and then checking out back to "8a0af6c" commit. Then I found that this commit doesn't make error but the recent one does.

I suspect "ff5fca7" commit's 
flightgoggles_uav_dynamics/CMakeLists.txt file's line 165 change as it added "multicopterDynamicsSim inertialMeasurementSim" target library as the error message says.

Here is my error message When I use recent commit.

Thanks Winter! Hope you are doing well.

mason@mason:~/catkin_ws$ catkin build
----------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/mason/catkin_ws
----------------------------------------------------------
Build Space:        [exists] /home/mason/catkin_ws/build
Devel Space:        [exists] /home/mason/catkin_ws/devel
Install Space:      [unused] /home/mason/catkin_ws/install
Log Space:         [missing] /home/mason/catkin_ws/logs
Source Space:       [exists] /home/mason/catkin_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
----------------------------------------------------------
[build] Found '9' packages in 0.0 seconds.                                                              
[build] Updating package table.                                                                         
Starting  >>> catkin_tools_prebuild                                                                     
Finished  <<< catkin_tools_prebuild                            [ 1.4 seconds ]                          
Starting  >>> flightgoggles_environment_creator                                                         
Starting  >>> flightgoggles_reporter                                                                    
Starting  >>> flightgoggles_uav_dynamics                                                                
Starting  >>> keyboard                                                                                  
Starting  >>> rosbaglive                                                                                
Finished  <<< rosbaglive                                       [ 1.5 seconds ]                          
Starting  >>> flightgoggles                                                                             
Finished  <<< flightgoggles_environment_creator                [ 1.5 seconds ]                          
________________________________________________________________________________________________________
Warnings   << keyboard:cmake /home/mason/catkin_ws/logs/keyboard/build.cmake.000.log                    
CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on '/usr/lib/x86_64-linux-gnu/libSDLmain.a' but
  neither '/usr/lib/x86_64-linux-gnu/libSDLmain.a_INCLUDE_DIRS' nor
  '/usr/lib/x86_64-linux-gnu/libSDLmain.a_LIBRARIES' is defined.
Call Stack (most recent call first):
[... 167 lines cut ...]
cd /home/mason/catkin_ws/build/keyboard; catkin build --get-env keyboard | catkin env -si  /usr/bin/cmake /home/mason/catkin_ws/src/keyboard --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mason/catkin_ws/devel/.private/keyboard -DCMAKE_INSTALL_PREFIX=/home/mason/catkin_ws/install; cd -
........................................................................................................
Finished  <<< flightgoggles_reporter                           [ 1.6 seconds ]                          
Finished  <<< keyboard                                         [ 3.3 seconds ]                          
Starting  >>> universal_teleop                                                                          
Finished  <<< flightgoggles_uav_dynamics                       [ 6.1 seconds ]                          
Finished  <<< universal_teleop                                 [ 4.3 seconds ]                          
Finished  <<< flightgoggles                                    [ 6.8 seconds ]                          
Starting  >>> flightgoggles_marker_visualizer                                                           
Starting  >>> flightgoggles_ros_bridge                                                                  
Finished  <<< flightgoggles_marker_visualizer                  [ 5.3 seconds ]                          
[... cut ...]
[build]   Warnings:  2 packages succeeded with warnings.                                                
[build]   Abandoned: None.                                                                              
[build]   Failed:    None.                                                                              
[build] Runtime: 23.9 seconds total.                                                                    
[build] Note: Workspace packages have changed, please re-source setup files to use them.