mit-aera/FlightGoggles #101: BUILD_BREAK: Latest FlightGoggle system does not build under catkin_make with custom code removed

SimpleUAVs, 2019-03-11:
Build still breaks at 97% Makefile:138 with all custom code removed after latest update

SimpleUAVs, 2019-03-11:
Problem appears to be in uav_dynamics;

/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp: In member function ‘void Uav_Dynamics::proceedState()’:
/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp:358:50: error: ‘dragFArmingThrustorce’ was not declared in this scope
                                  (2*c*d - 2*a*b)*dragFArmingThrustorce[1] +
                                                  ^
/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp:359:53: error: unable to find numeric literal operator ‘operator""ArmingThrust’
                                  (pow(a,2.) - pow(b,2.ArmingThrust) - pow(c,2.)
                                                     ^
/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp:359:53: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp:360:1: error: ‘ArmingThrust’ was not declared in this scope
 ArmingThrust
 ^
/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp: In member function ‘void Uav_Pid::controlUpdate(geometry_msgs::Vector3&, double*, double*, double*, double)’:
/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp:605:8: error: ‘sArmingThrust_t’ was not declared in this scope
   for (sArmingThrust_t ind = 0; ind < 3; ind++) {
        ^
/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp:605:33: error: ‘ind’ was not declared in this scope
   for (sArmingThrust_t ind = 0; ind < 3; ind++) {
                                 ^
/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp:606:5: error: ‘intSArmingThruste_’ was not declared in this scope
     intSArmingThruste_[ind] += dt * stateDev[ind];
     ^
/home/ags/catkin_ws/src/flightgoggles/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp:608:9: error: ‘ArmingThrust’ was not declared in this scope
     out[ArmingThrust] = propGain_[ind] * stateDev[ind] +
         ^
flightgoggles/flightgoggles_uav_dynamics/CMakeFiles/flightgoggles_uav_dynamics_node.dir/build.make:62: recipe for target 'flightgoggles/flightgoggles_uav_dynamics/CMakeFiles/flightgoggles_uav_dynamics_node.dir/src/flightgoggles_uav_dynamics_node.cpp.o' failed
make[2]: *** [flightgoggles/flightgoggles_uav_dynamics/CMakeFiles/flightgoggles_uav_dynamics_node.dir/src/flightgoggles_uav_dynamics_node.cpp.o] Error 1
CMakeFiles/Makefile2:3978: recipe for target 'flightgoggles/flightgoggles_uav_dynamics/CMakeFiles/flightgoggles_uav_dynamics_node.dir/all' failed
make[1]: *** [flightgoggles/flightgoggles_uav_dynamics/CMakeFiles/flightgoggles_uav_dynamics_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

Winter-Guerra, 2019-03-11:
Hi Jeff,

It seems like your code for the `flightgoggles_uav_dynamics` node has diverged from the version in the master branch. If you take a look at the source code output by your compiler, you will note that the compiler's source does not match [the source code in the FlightGoggles master branch](
https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles_uav_dynamics/src/flightgoggles_uav_dynamics_node.cpp). We suggest making sure that your local code matches the code in the FlightGoggles master branch. 

To verify that your source for the uav dynamics is the same as that in the FlightGoggles master branch, please run the following command:

```
cd ~/catkin_ws/src/flightgoggles 
git diff ca00b61b579f73cfd2420fb55d579884a88f6965 -- flightgoggles_uav_dynamics/src/
```  

Best,
-Winter

SimpleUAVs, 2019-03-11:
How did that happen. I’ve only done GIT PULLs daily, and for the last week the results have always been “Already up-to-date”, including with the bad file. 

 

Replacing the file manually and the build completes. Thanks for the TS.

 

jz

closed 2019-03-11 by Winter-Guerra