mit-aera/FlightGoggles #102: CHANGE_REQUEST: Split IMU from flight dynamics

SimpleUAVs, 2019-03-12:
flightgoggles_uav_dynamics publishes the /imu/data topic and uses the data published on this topic internally. The IMU data is simulated with significant artificial random noise. Therefore it is impossible to record this topic and the rateThrust topic in a rosbog file and play it back since 2 processes cannot publish to the same topic. If the rateThrust topic is played back only, the random noise in the /imu/data topic prevents accurate playback and over 50% failure in course completion using playback, pre-recorded control inputs as require for AI training.

Winter-Guerra, 2019-03-12:
Hi Jeff Zika,

The behavior you're seeing is expected. The IMU is modeled after a real-world sensor (Xsens - MTi 1). The dynamics simulator generates IMU measurements and does not expect to receive IMU measurements from any other source. 

The non-zero failure rate under open loop control is expected. If you would like to make open loop control inputs deterministic, simply lower the IMU and control input noise settings.

To regenerate images from an already recorded rosbag,  please take a look at [this example launchfile](https://github.com/mit-fast/FlightGoggles/blob/master/flightgoggles/launch/trajPlayback.launch). Note that this is an example launchfile, which may require changes for your particular use case.

Best,
-Winter

SimpleUAVs, 2019-03-13:
I understand this is by design, just asking if the IMU generator can be split off from UAV dynamics as a separate node.

Winter-Guerra, 2019-03-13:
Hi Jeff Zika,

Unfortunately, the generation of IMU measurements cannot be trivially decoupled from the rest of the physics simulation at this time. In the interest of fairness, we will not release any new features for FlightGoggles until the AlphaPilot challenge is concluded.

Best
-Winter

closed 2019-07-17 by Winter-Guerra