mit-aera/FlightGoggles #103: startup launch files cannot set starting position

SimpleUAVs, 2019-03-12:
The scorer.launch file sets the starting location to the start of the AlphaPilot course in flightgoggles, or used to. Neither the scorer.launch file or related yaml files have changed and attempt to set the correct location but this setting is ignored by flightgoggles_uav_dynamics. All launch files default to the center pose location no matter the individual starting poses.

The behavior changed in the last flightgoggles update.

Winter-Guerra, 2019-03-12:
Hey Jeff,

Could you please post your launch file terminal output and `rosparam list`? I suspect that you're having the same issue as #80, but cannot confirm without more info.

Best,
-Winter

SimpleUAVs, 2019-03-12:
[score.txt](https://github.com/mit-fast/FlightGoggles/files/2959125/score.txt)
 
The previous file shows the initial pose being set to the correct location in the challenge_final.yaml file but the initial pose actually set in the renderer (Unity) is the center of the course.

Winter-Guerra, 2019-03-13:
Hi Jeff Zika,
It looks like your `scorer.launch` file is missing the `<group ns="/uav">` tag around the `<rosparam ...>` tags. The fix for this issue has already been posted in a response to your previously authored issue (#80, step 2).
Best,
-Winter

SimpleUAVs, 2019-03-13:
This was downloaded as part of the challenge_leaderboard.zip file. Do you want to bug the AlphaPilot / DRL devs? Its rather hard to get a response on the herox site.

SimpleUAVs, 2019-03-13:
That seems to fix it. Thanks.

Winter-Guerra, 2019-03-13:
I'm glad it works! Closing.

closed 2019-03-13 by Winter-Guerra